lmarzen/esp32-weather-epd #1: Сhange design picture

Dozentz, 2022-11-23:
Tell me how to change desi
[image]
gn picture of the image, for example to your own

lmarzen, 2022-11-23:
Hi Dozentz, 
Are you trying to change the displayed image to look like the one you provided?
In that case, you would need to make changes to `renderer.cpp`.

Specifically, the following functions are called from `setup()` in `main.cpp` and would need to be altered to draw icons and text in the desired position.
`void drawCurrentConditions(owm_current_t &current, owm_daily_t &today,
                           owm_resp_air_pollution_t &owm_air_pollution, 
                           float inTemp, float inHumidity);`
`void drawForecast(owm_daily_t *const daily, tm timeInfo);`
`void drawAlerts(std::vector<owm_alerts_t> &alerts,
                const String &city, const String &date);`
`void drawLocationDate(const String &city, const String &date);`
`void drawOutlookGraph(owm_hourly_t *const hourly, tm timeInfo);`
`void drawStatusBar(String statusStr, String refreshTimeStr, int rssi, 
                   double batVoltage);`

Dozentz, 2022-11-23:
Is it possible to display this page as html ?

lmarzen, 2022-11-23:
Unfortunately, that is not possible with the current implementation, and I do not know how where to start with that. Maybe there is some library that would make adding that functionality relatively straightforward?

Dozentz, 2022-11-24:
There is a project, but it is implemented on Raspberry. You have a more budget option in terms of components
https://github.com/kimmobrunfeldt/eink-weather-display

lmarzen, 2022-11-24:
Looks interesting, good luck if you decide to add html support, I would gladly merge it into this project if you do so.

Dozentz, 2022-11-25:
I studied your project and realized that your execution is simpler and clearer. Tell me in more detail how to arrange the icons in the renderer.cpp file and is there any screen emulator for debugging, so as not to constantly reload the firmware in the ESP

lmarzen, 2022-11-26:
I did not use any screen emulator when I wrote the code; I reflashed it every time to test. 

The icons are all drawn using the `drawInvertedBitmap()` function. For example the big icon in the top left corner that displays the current weather condition is drawn by the following function call: `  display.drawInvertedBitmap(0, 0,
                             getCurrentConditionsBitmap196(current, today), 
                             196, 196, GxEPD_BLACK);`
Where 0, 0  is the coordinates at which the bitmap should be drawn. (0, 0  is the top left corner, 480, 800 is the bottom right corner). getCurrentConditionsBitmap196(current, today) is a helper function that returns the address of the bitmap, formatted in a c-style array. 196, 196 are the dimensions of the icon. And GxEPD_BLACK is the color, in which the image will be drawn.