lmarzen/esp32-weather-epd #3: Work from USB ?

Dozentz, 2022-12-20:
Will work from usb? I have a constantly blinking red battery light and a white screen

lmarzen, 2022-12-20:
Yes it will work over usb.If pressing the reset button doesn't cause the display to do anything then it may be wired incorrectly?

Dozentz, 2022-12-20:
Connected as picture
[image]
[image]

lmarzen, 2022-12-20:
The switches on the e-paper hat. "Interface Config: Set switch to position 1."

lmarzen, 2022-12-20:
You could also try disconnecting it from usb power and then reconnecting it. Sometimes the esp32s can be finicky after you upload your code.

Dozentz, 2022-12-20:
I turned off the power and turned it on again, set the flag to 1. Tell me what else can be done?

lmarzen, 2022-12-20:
Are you sure the code was successfully uploaded to the esp32?

lmarzen, 2022-12-20:
I'm running out of ideas; maybe a try to disconnect and reconnect the connectors?

Dozentz, 2022-12-21:
upload complete ?
[image]

lmarzen, 2022-12-21:
yup, that looks good:/

lmarzen, 2022-12-21:
open the serial monitor while the esp32 is connected via usb. It will print debug debug messages that indicate what it is doing.

Dozentz, 2022-12-21:
Tell me how to do it?

lmarzen, 2022-12-21:
in vscode at the bottom(near upload and build buttons) there should be a button to open the serial monitor. Hover over the buttons to find it. Once you find and click it, press the reset button on the esp32, then the esp32 should print out some debug messages that i put in the code to indicate what it is doing. If the esp32 is hanging somewhere this should give me enough information to figure out what might be causing it.

Dozentz, 2022-12-21:
debug
[image]
[image]

lmarzen, 2022-12-21:
Oh weird, looks to be having a boot error before it is executing any of my code. I can look into this, but I am traveling for the holidays and wont be able to try to recreate the issue until mid next month.

Dozentz, 2023-01-16:
Good afternoon. Were you able to fix the error?

lmarzen, 2023-01-16:
I forgot about this over holiday, I have some time today to work on it. I will keep you updated.

lmarzen, 2023-01-16:
I am able to recreate the issue you were describing simply by uploading my existing known good project. Some configuration thing must have changed. hmmmmm

lmarzen, 2023-01-16:
I found another group of people with the same issue. https://github.com/esphome/issues/issues/3921

lmarzen, 2023-01-16:
So I guess the issue is that the firebeetle v2 that I used for this project only has 4MB of flash, while other versions of the esp32 have more flash. I have no idea why it was working before if this is the issue. Sounds like some upload tool got smarter and that the firebeetle v2 might also not have proper support. I have no idea what has happened, I guess I will change my controller out for the older firebeetle that has 16MB of flash and micro usb.

Dozentz, 2023-01-16:
My controller version is type С. How can I find out how much flash memory is in my version?

lmarzen, 2023-01-16:
https://www.dfrobot.com/product-2195.html vs https://www.dfrobot.com/product-1590.html

lmarzen, 2023-01-16:
The usb C version is only 4MB. Somehow this never caused an issue until now, even though the compiled code is like 11MB.

Dozentz, 2023-01-16:
I understand the problem is now with the compiler

lmarzen, 2023-01-16:
No my understanding is that this board is properly supported in platformio and that when I originally created this project for some reason I was able to upload a bigger binary to the controller without it freaking out, but now there is some extra check or maybe even just a change in environment that is causing this issues. I have no idea why this worked in the first place with a 4MB flash. Clearly 11MB > 4MB. I must have over looked that the newer usb c firebeetle had less flash than its predecessor. So the solution is that you must buy the micro usb firebeetle because that has 16mb of flash. I am testing it now because I have the hardware on hand.

lmarzen, 2023-01-16:
Okay, I have dug more into it. It seems to be a support issue with platformio. Maybe not flash size issue?

Dozentz, 2023-01-16:
Need to ask on the forum https://community.platformio.org.

lmarzen, 2023-01-16:
Alright, forget my earlier explanation. I think I got it now. So the Firebeetle v2 esp32-e is not official supported but it worked earlier last year. So the solution is to use an older version of expressif32, So in platform.ini try change `platform = espressif32` to `platform = espressif32@5.0.0`

lmarzen, 2023-01-16:
https://github.com/platformio/platform-espressif32/issues/644

lmarzen, 2023-01-17:
Alright. Fingers crossed, the issue should be fixed now. Let me know how it goes.

Dozentz, 2023-01-17:
Ok

Dozentz, 2023-01-18:
Everything worked. Great job. Tell me how to make a weather icon shadow with curves and a battery with 4 compartments. Thanks to)
[image]

closed 2023-01-18 by lmarzen

lmarzen, 2023-01-18:
I created a new issue with instructions for changing icons. https://github.com/lmarzen/esp32-weather-epd/issues/4