lmarzen/esp32-weather-epd #6: Cyrillic Font Support

lmarzen, 2023-01-27:
Great, everything works. How to add Cyrillic support in a font? Does freesans support by default cyrillic.

_Originally posted by @Dozentz in https://github.com/lmarzen/esp32-weather-epd/issues/5#issuecomment-1406905035_

lmarzen, 2023-01-27:
@Dozentz It appears that FreeSans has Cyrillic characters in it (https://www.gnu.org/software/freefont/), but the tool(https://github.com/adafruit/Adafruit-GFX-Library/tree/master/fontconvert) that I am using to generate the c-code with the font data says this...
`Currently this only extracts the printable 7-bit ASCII chars of a font.

So one you can add Cyrillic characters by replacing some of the following Latin characters, (this might be a little messy)
```
0x20 ' '
0x21 '!'
0x22 '"'
0x23 '#'
0x24 '$'
0x25 '%'
0x26 '&'
0x27 '''
0x28 '('
0x29 ')'
0x2A '*'
0x2B '+'
0x2C ','
0x2D '-'
0x2E '.'
0x2F '/'
0x30 '0'
0x31 '1'
0x32 '2'
0x33 '3'
0x34 '4'
0x35 '5'
0x36 '6'
0x37 '7'
0x38 '8'
0x39 '9'
0x3A ':'
0x3B ';'
0x3C '<'
0x3D '='
0x3E '>'
0x3F '?'
0x40 '@'
0x41 'A'
0x42 'B'
0x43 'C'
0x44 'D'
0x45 'E'
0x46 'F'
0x47 'G'
0x48 'H'
0x49 'I'
0x4A 'J'
0x4B 'K'
0x4C 'L'
0x4D 'M'
0x4E 'N'
0x4F 'O'
0x50 'P'
0x51 'Q'
0x52 'R'
0x54 'T'
0x55 'U'
0x56 'V'
0x57 'W'
0x58 'X'
0x59 'Y'
0x5A 'Z'
0x5B '['
0x5C '\'
0x5D ']'
0x5E '^'
0x5F '_'
0x60 '`'
0x61 'a'
0x62 'b'
0x63 'c'
0x64 'd'
0x65 'e'
0x66 'f'
0x67 'g'
0x68 'h'
0x69 'i'
0x6A 'j'
0x6B 'k'
0x6C 'l'
0x6D 'm'
0x6E 'n'
0x6F 'o'
0x70 'p'
0x71 'q'
0x72 'r'
0x73 's'
0x74 't'
0x75 'u'
0x76 'v'
0x77 'w'
0x78 'x'
0x79 'y'
0x7A 'z'
0x7B '{'
0x7C '|'
0x7D '}'
0x7E '~'
```

Dozentz, 2023-01-27:
change in this file FreeSans4pt7b.h ?
`const GFXglyph FreeSans4pt7bGlyphs[] PROGMEM = {
  {     1,   1,   6,   3,    1,   -5 },   // 0x21 'my symbol'
`

lmarzen, 2023-01-27:
Not quite. You would need to edit the .ttf file. See the README in the fonts folder for more information about regenerating the fonts.

We should actually use the remaining bit to give use an additional 128 characters. I was looking at the font generation script and with some small changes this shouldn't be too hard.

lmarzen, 2023-01-27:
I will do some experimenting later today(hoping I will have some time) to see if I can make any progress.

lmarzen, 2023-01-28:
[image]

Okay, I figured it out. It isn't too bad.

My solution is to use the 8th bit for characters in order to add 128 new Cyrillic characters. I wrote a script called ttf_to_adafruit_gfx.sh, which can assist in remapping a subset of the Cyrillic Unicode characters to the extended ASCII set. See the README in the fonts folder for directions on how to run it.

I created a branch so that you can download my changes and test it out...
https://github.com/lmarzen/esp32-weather-epd/tree/cyrillic-alphabet-test

I simply mapped unicode characters U0400-U047F to ASCII 0x80-0xFF. This is done in the for loop of ttf_to_adafruit_gfx.sh starting on line 29. (incase you need to know where to start if you want to make a more meaningful subset).
https://www.unicode.org/charts/PDF/U0400.pdf

Now the string you see in the image I shared is the result of this code in config.cpp:
`const String CITY_STRING = "\x80\x81\x82\x83\x84\x85\x86\x87\x88\x89\x8A\x8B\x8C\x8D\x8E\x8F\x90\x91\x92\x93\x94\x95\x96\xFF";`

As you can see if you refer to the unicode chart linked above, \x80 corresponds with U0400 (Ѐ), \x81 corresponds with U0401 (А), etc...

Dozentz, 2023-01-28:
Thanks to. Everything works. There is also a small request to add icons of size 220 and 256. You will get it faster. Thank you

lmarzen, 2023-01-28:
You're welcome. I am very glad to hear that!

For generating bigger fonts see the README in the fonts folder, you will need to use the script I mentioned before and will need to make some changes at the top of the script that indicate what sizes you want. Then you will need to copy the generated fonts to platformio/src/fonts then you need to add an #include statement at the top of the file that you wish to use one of the new fonts. (ex: a 56pt temperature font would be included as follows `#include fonts/FreeSans56pt_temperature`) Then you can edit or add display.setFont(&FreeSans56pt_temperature); as you please.

If you wanted to make the main temperature font bigger, then you would go to line 232 in renderer.cpp and replace "FreeSans48pt_temperature" with "FreeSans56pt_temperature". (don't forget to add the #include at the top of the file)

For Icons, it is similar to fonts. go to the icons folder and read the README. This will tell you more about what you need to do in order to generate new icons of different sizes. Then as you did with the fonts, copy the icons to platformio/src/icons, add the #include statement and make any changes to the code that you want.

The hardest part here (other than maybe getting the scripts working on your end; they are intended for Linux environments) is probably going to be the trial and error to get the icons and fonts in the desired place. Admittedly this was a pain when I first was programming this project, it takes so long to recompile and upload. Every time platformio checks for updates to dependencies, ugh.

Good luck, and let me know if you have any other questions.

closed 2023-01-28 by lmarzen