lmarzen/esp32-weather-epd #11: One Call deserializeJson() failed: IncompleteInput

nglessner, 2023-02-10:
I am able to connect to wifi, pull down the time from the NTP server, but then I get an error "One Call deserializeJson() failed: IncompleteInput"

This repeats 3x before continuing on.  I can confirm that in [Open Weather API Subscriptions](https://home.openweathermap.org/subscriptions) that I see the call attempts increase.  I am following this project verbatim using a DFRobot Firebeetle 2 (ESP32-E) and a BME-280.

lmarzen, 2023-02-10:
Hi nglessner, I will look into this and see if I can reproduce it.  When you have time, could you please provide **sanitized** copies of your config.cpp and config.h? (remove wifi ssid/password, api key) You may also want to obfuscate your lat long. If you decide to obfuscate your location, please make sure that you still get the same error for the new lat/long.

Thanks, Luke.

lmarzen, 2023-02-10:
https://arduinojson.org/v6/api/misc/deserializationerror/#deserializationerrorincompleteinput
> The end of the input is missing. Possible reasons:
>    - a buffer was too small to contains the input
>    - a timeout occurred when reading from a stream

Hopefully increasing one of these 2 things can fix resolve this issue.

nglessner, 2023-02-10:
I'm guessing it'll be the second one. I was looking at the first possible cause, but I'm seeing my free heap go down by just over 32k which is what I'd expect. 

Heap: 148955
Heap: 116175
One Call deserializeJson() failed: IncompleteInput
Heap: 147011
Heap: 114231
One Call deserializeJson() failed: IncompleteInput
Heap: 148731
Heap: 115951
One Call deserializeJson() failed: IncompleteInput

nglessner, 2023-02-10:
Here's my config:

#include <vector>
#include <Arduino.h>
#include "config.h"

// PINS
// ADC pin used to measure battery voltage
const uint8_t PIN_BAT_ADC  = A2;
// Pins for Waveshare e-paper Driver Board
const uint8_t PIN_EPD_BUSY = 13;
const uint8_t PIN_EPD_CS   =  2;
const uint8_t PIN_EPD_RST  = 21;
const uint8_t PIN_EPD_DC   = 22;
const uint8_t PIN_EPD_SCK  = 18;
const uint8_t PIN_EPD_MISO = 19; // 19 Master-In Slave-Out not used, as no data from display
const uint8_t PIN_EPD_MOSI = 23;
// I2C Pins used for BME280
const uint8_t PIN_BME_SDA = 17;
const uint8_t PIN_BME_SCL = 16;
const uint8_t BME_ADDRESS = 0x76; // if sensor does not work, try 0x77

// WIFI CREDENTIALS
const char *WIFI_SSID     = "XXXXX";
const char *WIFI_PASSWORD = "XXXXX";

// OPENWEATHERMAP API
// OpenWeatherMap API key, https://openweathermap.org/
const String OWM_APIKEY   = "XXXXX";
const String OWM_ENDPOINT = "api.openweathermap.org";
// OpenWeatherMap One Call 2.5 API is deprecated for all new free users 
// (accounts created after Summer 2022).
//
// Please note, that One Call API 3.0 is included in the "One Call by Call" 
// subscription only. This separate subscription includes 1,000 calls/day for 
// free and allows you to pay only for the number of API calls made to this 
// product.
//
// Here’s how to subscribe and avoid any credit card changes:
// - Go to https://home.openweathermap.org/subscriptions/billing_info/onecall_30/base?key=base&service=onecall_30
// - Follow the instructions to complete the subscription.
// - Go to https://home.openweathermap.org/subscriptions and set the "Calls per
//   day (no more than)" to 1,000. This ensures you will never overrun the free 
//   calls.
const String OWM_ONECALL_VERSION = "3.0";

// LOCATION
// Set your latitude and longitude.
// (used to get weather data as part of API requests to OpenWeatherMap)
const String LAT = "39.6910";
const String LON = "-89.7105";
// City name that will be shown in the top-right corner of the display.
const String CITY_STRING = "City, State";

// TIME
// For list of time zones see 
// https://github.com/nayarsystems/posix_tz_db/blob/master/zones.csv
const char *TIMEZONE = "CST6CDT,M3.2.0,M11.1.0";
// Time format used when displaying sunrise/set times. (Max 11 characters)
// For more information about formatting see
// https://man7.org/linux/man-pages/man3/strftime.3.html
// const char *TIME_FORMAT = "%l:%M%P"; // 12-hour ex: 1:23am  11:00pm
const char *TIME_FORMAT = "%l:%M%P";   // 24-hour ex: 01:23   23:00
// Time format used when displaying axis labels. (Max 11 characters)
// For more information about formatting see
// https://man7.org/linux/man-pages/man3/strftime.3.html
// const char *HOUR_FORMAT = "%l%P"; // 12-hour ex: 1am  11pm
const char *HOUR_FORMAT = "%l%P";      // 24-hour ex: 01   23
// Date format used when displaying date in top-right corner. For more 
// For more information about formatting see
// https://man7.org/linux/man-pages/man3/strftime.3.html
const char *DATE_FORMAT = "%A, %B %e"; // Saturday, January 1
// Date/Time format used when displaying the last refresh time along the bottom
// of the screen
// For more information about formatting see
// https://man7.org/linux/man-pages/man3/strftime.3.html
const char *REFRESH_TIME_FORMAT = "%x %H:%M";
// NTP_SERVER_1 is the primary time server, while NTP_SERVER_2 is a fallback.
// In most cases it's best to use pool.ntp.org to find an NTP server
// The system will try finding the closest available servers for you.
const char *NTP_SERVER_1 = "us.pool.ntp.org";
const char *NTP_SERVER_2 = "time.nist.gov";
// Sleep duration in minutes. (aka how often esp32 will wake for an update)
// Aligned to the nearest minute boundary, so if 30 will always update at 00 or 
// 30 past the hour.
const long SLEEP_DURATION = 10;
// If BED_TIME == WAKE_TIME, then this battery saving feature will be disabled.
const int BED_TIME  = 00; // Last update at 00:00 (midnight) until WAKE_TIME.
const int WAKE_TIME = 05; // Hour of first update after BED_TIME, 06:00.

// HOURLY OUTLOOK GRAPH
// Number of hours to display on the outlook graph.
// Value must be between 8-48 (inclusively).
const int HOURLY_GRAPH_MAX = 24;

// BATTERY
// To protect the battery upon LOW_BATTERY_VOLTAGE, the display will cease to
// update until battery is charged again. The ESP32 will deep-sleep (consuming
// < 11μA), waking briefly check the voltage at the corresponding interval (in
// minutes). Once the battery voltage has fallen to CRIT_LOW_BATTERY_VOLTAGE,
// the esp32 will hibernate and a manual press of the reset (RST) button to
// begin operating again.
const float LOW_BATTERY_VOLTAGE      = 3.20; // (volts)
const float VERY_LOW_BATTERY_VOLTAGE = 3.10; // (volts)
const float CRIT_LOW_BATTERY_VOLTAGE = 3.00; // (volts)
const unsigned long LOW_BATTERY_SLEEP_INTERVAL      = 30;  // (minutes)
const unsigned long VERY_LOW_BATTERY_SLEEP_INTERVAL = 120; // (minutes)

// LANGUAGE
// see config.h

// UNITS
// see config.h

// AIR QUALITY INDEX
// see config.h

// WIND
// see config.h

lmarzen, 2023-02-10:
Thank you. I will see if I can reproduce your issue later this evening.

lmarzen, 2023-02-11:
I was not able to reproduce the issue, but I think is possible that it could have been buffer size. I actually have some api responses that I saved as examples when I was experimenting months ago when I was first programming this project, one of the responses I saved had a size of 35.2kb. I increased the buffer size from 32kb to 48kb. Not sure if that will fix your issue.

lmarzen, 2023-02-11:
You can try increasing the timeout to 10s (default 1s) by adding `json.setTimeout(10000);` to line 9 of api_response.cpp at the beginning of deserializeOneCall().

nglessner, 2023-02-11:
What I'm seeing is very slow incoming data from api.openweathermap.org, but I think the speed issue is related to my MCU. Is the platformio.ini in main up to date?  

I used the suggested StreamUtils package to output the wifi client stream to serial and I would see the data arrive very slowly in chunks.  After updating the platformio.ini to use the latest platform and lib_deps I would get a significantly faster wifi connection (both establishing connection and data incoming), but I would also get this I2C error:

[esp32-hal-i2c-slave.c:224] i2cSlaveInit(): invalid pins sda=16, scl=-96
[  3411][E][Wire.cpp:167] begin(): Slave Init ERROR

I'm still fiddling with the library versions trying to figure out which combination will work for me.  If I have it set to the config below, it will give me the IncompleteInput error.  If I change the platform = espressif32 @ 4.4.0, wifi will run fast but I will get the i2cSlaveInit error.

[env:firebeetle32]
platform = espressif32 @ 4.0.0
board = firebeetle32
framework = arduino
lib_deps = 
	bblanchon/ArduinoJson
	zinggjm/GxEPD2 @ ^1.4.5
	adafruit/Adafruit Unified Sensor @ ^1.1.5
	adafruit/Adafruit BusIO @ ^1.11.2
	adafruit/Adafruit BME280 Library @ ^2.2.2
	bblanchon/StreamUtils@^1.7.0
monitor_speed = 115200
board_build.f_cpu = 160000000L

lmarzen, 2023-02-11:
There is an open issue https://github.com/platformio/platform-espressif32/issues/644 to add official support for the firebeetle esp32-e into platformio. Until then, it seems we are stuck on this older version (4.0.0.) 5.1 won't even let you upload to the code, you get a bootloader error. I too saw I2C errors after versions 4.0.

lmarzen, 2023-02-11:
To give you more insight, I only recently added @ 4.0.0 to the .ini file. When I originally built this in the summer of 2022 I did not specify the version of platform espressif32, (presumably I was using the latest stable version). Then when I tried to rebuild the same code again in December of 2022 I discovered that the new versions of espressif32 broke compatibility with the firebeetle esp32-e. I had no idea this board wasn't officially supported yet.

lmarzen, 2023-02-11:
I found the change that I suspect broke the esp32-e's compatibility; these changes in firebeetle32.json. https://github.com/platformio/platform-espressif32/commit/cb3583813d0dad75edc0853cfec865c89b221263. I can experiment some tomorrow. I'm tempted to create a pull request to add a firebeetle32e board for platformio.

lmarzen, 2023-02-11:
In order to upload code to the Firebeetle 2 ESP32-E using the latest version of espressif32, we can override 3 of the default settings for the firebeetle32 board configuration. See my platformio.ini file below. I already updated the repo with these changes (https://github.com/lmarzen/esp32-weather-epd/commit/5bed202f4725047fa3a9c57d04c9c60c8765b069). I immediately noticed that my esp32-e connects to wifi much faster on the latest espressif32. This makes me hopeful that we may have solved your problem.
```
[env:firebeetle32]
platform = espressif32
board = firebeetle32
framework = arduino

; override default settings to enable compatibility with Firebeetle 2 ESP32-E
board_upload.flash_size = 4MB
board_upload.maximum_ram_size = 327680
board_upload.maximum_size = 4194304
```

nglessner, 2023-02-11:
This worked! What pointed you towards fixing the flash size?  I'm also curious what made you undo the change to fix the ram size? DFRobots spec says 520KB, but it looks like I'm limited to the 320KB that was built into the original ESP32 spec.

lmarzen, 2023-02-11:
When you try to upload to the esp32-e with the latest platform = espressif32, it gives an error about the flash size being incorrect. This was my first indication. Since I knew that the older versions of espressif32 worked, I looked into what had changed for the firebeetle32 board definition. I found this specific commit that I suspect broke the compatibility, https://github.com/platformio/platform-espressif32/commit/cb3583813d0dad75edc0853cfec865c89b221263.

I reverted the explicit override for ram size because the "firebeetle32" board definition was already using 520KB. So my override to 520KB was pointless. If what you are saying is true, do you think we need to limit it to 320KB?

I also have moved to add official support for the firebeetle 2 esp32-e to arduino-esp32 and platformio. (https://github.com/espressif/arduino-esp32/pull/7835 and https://github.com/platformio/platform-espressif32/pull/1026)

closed 2023-02-13 by lmarzen