lmarzen/esp32-weather-epd #12: "Unexpected error occured" during build

rknobbe, 2023-02-11:
I'm trying to build from the github main branch.  Windows 11, Visual Studio Code  Version 1.75.1, PlatformIO Core 6.1.6, Home 3.4.3.  Build fails with the following trackback.  I've uninstalled and reinstalled VSCode and PlatformIO with no change.  I'm not an expert in either, so this could very well be a problem behind the keyboard.  I was able to compile a simple Hello World in VSCode.

Any pointers?

`Resolving firebeetle32 dependencies...
Error: Traceback (most recent call last):
  File "C:\Users\roger\.platformio\penv\lib\site-packages\platformio\__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\Users\roger\.platformio\penv\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\roger\.platformio\penv\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\Users\roger\.platformio\penv\lib\site-packages\platformio\cli.py", line 71, in invoke
    return super().invoke(ctx)
  File "C:\Users\roger\.platformio\penv\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\roger\.platformio\penv\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\roger\.platformio\penv\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\roger\.platformio\penv\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\roger\.platformio\penv\lib\site-packages\platformio\project\commands\init.py", line 94, in project_init_cmd
    install_project_dependencies(
  File "C:\Users\roger\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 106, in install_project_dependencies
    already_up_to_date = not install_project_env_dependencies(env, options)
  File "C:\Users\roger\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 132, in install_project_env_dependencies
    _install_project_env_libraries(project_env, options),
  File "C:\Users\roger\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 247, in _install_project_env_libraries
    env_lm.install(
  File "C:\Users\roger\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 48, in install
    pkg = self._install(spec, skip_dependencies=skip_dependencies, force=force)
  File "C:\Users\roger\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 92, in _install
    self.install_dependencies(pkg, print_header=False)
  File "C:\Users\roger\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 140, in install_dependencies
    self.install_dependency(dependency)
  File "C:\Users\roger\.platformio\penv\lib\site-packages\platformio\package\manager\library.py", line 89, in install_dependency
    not_builtin_conds.append(not self.is_builtin_lib(spec.name))
  File "C:\Users\roger\.platformio\penv\lib\site-packages\platformio\package\manager\library.py", line 120, in is_builtin_lib
    for storage in cls.get_builtin_libs():
  File "C:\Users\roger\.platformio\penv\lib\site-packages\platformio\util.py", line 56, in wrapper
    self.cache[key] = (time.time(), func(*args, **kwargs))
  File "C:\Users\roger\.platformio\penv\lib\site-packages\platformio\package\manager\library.py", line 105, in get_builtin_libs
    for storage in p.get_lib_storages():
  File "C:\Users\roger\.platformio\penv\lib\site-packages\platformio\platform\base.py", line 216, in get_lib_storages
    pkg = self.get_package(opts["package"])
  File "C:\Users\roger\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 29, in get_package
    return self.pm.get_package(spec or self.get_package_spec(name))
  File "C:\Users\roger\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 21, in get_package_spec
    owner=self.packages[name].get("owner"),
KeyError: 'framework-stm32cube'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

`

lmarzen, 2023-02-11:
Hmmm. I haven't seen this one. A quick search found this https://community.platformio.org/t/platformio-has-stopped-working/30605

rknobbe, 2023-02-12:
Not sure how but uninstalling VSCode and PIO, then deleting .platformio seemed to clear it up. Thanks for checking!

closed 2023-02-12 by rknobbe