lmarzen/esp32-weather-epd #15: Receive error when uploading to FireBeetle from USB

NWstudent, 2023-02-28:
....
.....
Uploading .pio/build/firebeetle32/firmware.bin
esptool.py v4.4
Serial port /dev/cu.Bluetooth-Incoming-Port
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
*** [upload] Error 2

lmarzen, 2023-03-01:
This is likely an issue with drivers. You would need to share more information if you want help diagnosing this. Check out the troubleshooting link that is in the output you shared. This issue does not seem to be uncommon. Try googling the error. There are many conversations about it online with steps you can take to resolve it.

closed 2023-03-05 by lmarzen

NWstudent, 2023-03-07:
I have attempted to install both recommend drivers for Mac (Apple M1 Max) OS (Ventura-13.2.1). Have attempted to install using VScode and Arduino IDE. I have attempted with 2 FireBeetle boards (p/n:DFR0654), two cables, two usb-c ports to rule out hardware malfunction. I also attempted to upload to a wired board and a clean board to determine if it had to be built out prior to upload. Each time the process gets the point where it performs:

Serial port /dev/cu.Bluetooth-Incoming-Port
Connecting......................................

and then shows:

A fatal error occurred: Failed to connect to ESP32: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
*** [upload] Error 2
=================================== [FAILED] Took 44.78 seconds ===================================

 *  The terminal process "platformio 'run', '--target', 'upload'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

The same exact process happens if I attempt to run the upload with nothing attached to the computer. Once the recommended drivers were installed the Mac recognizes the board as a USB Serial:

  Product ID:	0x7522
  Vendor ID:	0x1a86
  Version:	2.64
  Speed:	Up to 12 Mb/s
  Location ID:	0x02100000 / 1
  Current Available (mA):	500
  Extra Operating Current (mA):	0

Full output of the install is:

Executing task: platformio run --target upload 

Processing firebeetle32 (platform: espressif32; board: firebeetle32; framework: arduino)
----------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/firebeetle32.html
PLATFORM: Espressif 32 (6.0.1) > FireBeetle-ESP32
HARDWARE: ESP32 80MHz, 520KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20006.221224 (2.0.6) 
 - tool-esptoolpy @ 1.40400.0 (4.4.0) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 40 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 6.20.1
|-- GxEPD2 @ 1.5.0
|   |-- Adafruit GFX Library @ 1.11.5
|   |   |-- Adafruit BusIO @ 1.14.1
|   |   |   |-- Wire @ 2.0.0
|   |   |   |-- SPI @ 2.0.0
|   |   |-- Wire @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- Adafruit BusIO @ 1.14.1
|   |   |-- Wire @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|   |-- SPI @ 2.0.0
|-- Adafruit Unified Sensor @ 1.1.7
|-- Adafruit BusIO @ 1.14.1
|   |-- Wire @ 2.0.0
|   |-- SPI @ 2.0.0
|-- Adafruit BME280 Library @ 2.2.2
|   |-- Adafruit BusIO @ 1.14.1
|   |   |-- Wire @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|   |-- SPI @ 2.0.0
|   |-- Adafruit Unified Sensor @ 1.1.7
|-- pollutant-concentration-to-aqi
|-- HTTPClient @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- WiFiClientSecure @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|-- SPI @ 2.0.0
|-- WiFi @ 2.0.0
|-- Preferences @ 2.0.0
|-- Wire @ 2.0.0
Building in release mode
Retrieving maximum program size .pio/build/firebeetle32/firmware.elf
Checking size .pio/build/firebeetle32/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  18.9% (used 100500 bytes from 532480 bytes)
Flash: [========= ]  92.9% (used 1217337 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: /dev/cu.Bluetooth-Incoming-Port
Uploading .pio/build/firebeetle32/firmware.bin
esptool.py v4.4
Serial port /dev/cu.Bluetooth-Incoming-Port
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
*** [upload] Error 2
=================================== [FAILED] Took 44.90 seconds ===================================

 *  The terminal process "platformio 'run', '--target', 'upload'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

lmarzen, 2023-03-07:
This issue isn't specific to this project.  I have not encountered this issue myself, my only meaningful suggestion is to try what is mentioned on other forums. It sounds like you are on the right track though. I would disconnect everything from the esp32 when troubleshooting this issue, as it is a known issue that an incorrectly wired spi bus can cause some connection issues. I would also try another cable and different ports on your computer. Maybe even try on a different computer if you have one available.