lmarzen/esp32-weather-epd #17: Change charakter set

mathiasm75, 2023-03-04:
Hi first of all, great project! Thank you. Also for absolute newbie like me, i suceeded!

I made a translation into german, but can I also change the charakter set? I need the characters "ä", "ö" ...

Thank you, 
Mathias from Austria

lmarzen, 2023-03-04:
Thank you, I am glad to hear you were able to follow the documentation.

To address your question, yes. We can change the character set. The current state of this project has some limitations in this regard. It isn't the most user-friendly as currently the display only has the 7-bit ASCII character set. The solution I have used in the past is described here (https://github.com/lmarzen/esp32-weather-epd/issues/6). It entails using the unused 8th bit of ASCII to extend the character set and add 128 new characters. In the future, I want to research adding Unicode support, not sure how challenging this will be, but it seems like the ideal solution.

I am pretty busy this next week so I may not be able to provide much assistance until March 11. 

-Luke

aggr0lite, 2023-03-05:
Just looking into the same functionality, what a surprise! Just installed everything and then I was realizing Umlaut characters are not supported. 

I was looking at another (similar) project as well which supports Umlaut characters, but I like your dashboard better as it's cleaner, great work!

So the project I was looking at first is this one: https://github.com/G6EJD/ESP32-e-Paper-Weather-Display
It uses this library for Umlaut characters: https://github.com/olikraus/u8g2

Unfortunately I don't have any idea how to integrate it myself but it works without issues.

Keep up your great work, thanks for uploading this!

lmarzen, 2023-03-05:
I see. So [u8g2](https://github.com/olikraus/u8g2) takes the approach of creating custom character sets each limited to 256 characters.

lmarzen, 2023-03-05:
For now I added the extended ascii character set (5939ad9f699a39d6c38fecee71f03bd5f23b121c). See https://www.ascii-code.com/.

Here is an example: "Test: \xE4\xEB"
[image]

aggr0lite, 2023-03-05:
That was a quick one. Looks great, I'll test and give feedback. Thanks!

aggr0lite, 2023-03-05:
Works like a charm! Thanks for fixing this so quickly. 
Do you still plan to integrate full support?

lmarzen, 2023-03-05:
Glad to hear.

I would like to have better support for more character sets. I am considering adding the u8g2 library you suggested. Alternatively, I think proper utf-8 support would be ideal, but due to flash storage constraints, this may unfortunately not be feasible.

Until I get a chance to evaluate this further I will leave this issue open.

lmarzen, 2023-03-10:
Update: I looked into modifying the adafruit-gfx library to add utf-8 support. It seems this will not work due to flash size limitations. Even just the Basic Multilingual Plane (https://en.wikipedia.org/wiki/Plane_(Unicode)) would result in a size of ~13.9MB for only the largest font size needed.

Possible paths forward:
1. It may be possible to compress the font bit map. (Complexity unknown, will compression be effective enough?)
2. FreeSans.otf is less than 1MB. We could store the file on the esp32 and generate the characters as needed. (Complexity unknown and may run slowly on esp32.)
3. Use u8g2 or a similar library. (Advantage of simplicity. Disadvantage of limited character sets)

aggr0lite, 2023-03-10:
I appreciate you're still looking into this. For me your latest addition with the ascii character set works perfectly fine. Are there any lmitiations with it?

lmarzen, 2023-03-10:
The main limitation is that you can only use the set of 255 characters in the extended ascii character set. Unicode, more specifically utf-8 would enable the use of many more characters.

ZYL9, 2024-05-27:
Thanks for your great work!
I've use the library "olikraus/ u8g2_for_adafruit_gFx.@ ^1.8.0" and just make a few tweaks in renderer.cpp to make it work.
I have successfully displayed the Chinese characters on the screen, and the other unicode characters should be simple. 
Hope you can consider adding related features when you are free.

Also, for reference, I used uncompressed Chinese fonts in two sizes (12pt/16pt), which worked well on FireBeetle 2 ESP32-E, here's the flash usage info when building.
RAM: [===] 31.3% (used 102596 bytes from 327680 bytes)
Flash: [======] 60.6% (used 1905669 bytes from 3145728 bytes)
And if you can crop the font library, I believe this can get much better flash usage.

Nasamars, 2024-12-17:
> Thanks for your great work! I've use the library "olikraus/ u8g2_for_adafruit_gFx.@ ^1.8.0" and just make a few tweaks in renderer.cpp to make it work. I have successfully displayed the Chinese [...]

Could you provide methods and code for successfully displaying Chinese characters? thanks！

whtech, 2025-03-13:
> Could you provide methods and code for successfully displaying Chinese characters? thanks！

**step 1:** 
 add   `olikraus/u8g2_for_adafruit_gFx @ ^1.8.0` dependency in platformio.ini

**step 2:**
add following two lines in renderer.cpp:
```
#include <U8g2_for_Adafruit_GFX.h>
U8G2_FOR_ADAFRUIT_GFX u8g2_for_adafruit_gfx;
```

**step 3:**
add two functions in renderer.cpp:

```
/* Draws a Chinese string with alignment
 */
void drawChineseString(int16_t x, int16_t y, const String &text, alignment_t alignment,
  uint16_t color)
{
display.setFont(&FONT_12pt8b);
u8g2_for_adafruit_gfx.begin(display); // connect u8g2 procedures to Adafruit GFX

int16_t x1, y1;
uint16_t w, h;
display.setTextColor(color);
display.getTextBounds(text, x, y, &x1, &y1, &w, &h);
if (alignment == RIGHT)
{
x = x - w;
}
if (alignment == CENTER)
{
x = x - w / 2;
}
display.setCursor(x, y);

u8g2_for_adafruit_gfx.setFontMode(0);            // use u8g2 transparent mode (this is default)
u8g2_for_adafruit_gfx.setFontDirection(0);       // left to right (this is default)
u8g2_for_adafruit_gfx.setForegroundColor(color); // apply Adafruit GFX color
u8g2_for_adafruit_gfx.setBackgroundColor(GxEPD_WHITE);  // apply Adafruit GFX color
u8g2_for_adafruit_gfx.setFont(u8g2_font_wqy16_t_gb2312); // select u8g2 font from here: https://github.com/olikraus/u8g2/wiki/fntlistall
u8g2_for_adafruit_gfx.setCursor(x, y);                   // start writing at this position

u8g2_for_adafruit_gfx.print(text.c_str());

return;
} // end drawChineseString

bool hasUnicode(const String &text)
{
  bool bHasUnicode = false;
  for (int i = 0; i < text.length(); i++)
  {
    char currentChar = text.charAt(i);
    // //Serial.print(currentChar);
    // if (currentChar > 127)
    // {
    //   bHasUnicode = true;
    //   break; // Stop checking once a Unicode character is found
    // }
    if (currentChar >= 0xC0 && currentChar <= 0xDF) {
      // Start of a two-byte UTF-8 character
      Serial.print("2-byte UTF-8: ");
      Serial.println(currentChar);
      bHasUnicode = true;
      break;
    } else if (currentChar >= 0xE0 && currentChar <= 0xEF) {
      // Start of a three-byte UTF-8 character
      Serial.print("3-byte UTF-8: ");
      Serial.println(currentChar);
      bHasUnicode = true;
      break;
    } else if (currentChar >= 0xF0 && currentChar <= 0xF7) {
      // Start of a four-byte UTF-8 character
      Serial.print("4-byte UTF-8: ");
      Serial.println(currentChar);
      bHasUnicode = true;
      break;
    }
  }
  return bHasUnicode;
}
```

**step4:**
add definition in renderer.h: 

```
void drawChineseString(int16_t x, int16_t y, const String &text, alignment_t alignment,
                uint16_t color=GxEPD_BLACK);
bool hasUnicode(const String &text);
```

**step 5:**
 modify the function displayString in renderer.cpp, focus on the first four lines: 

```
/* Draws a string with alignment
 */
void drawString(int16_t x, int16_t y, const String &text, alignment_t alignment,
                uint16_t color)
{
  if(hasUnicode(text)){
    Serial.print("has Chinese");
    drawChineseString(x, y, text, alignment, color);
  } else {
  int16_t x1, y1;
  uint16_t w, h;
  display.setTextColor(color);
  display.getTextBounds(text, x, y, &x1, &y1, &w, &h);
  if (alignment == RIGHT)
  {
    x = x - w;
  }
  if (alignment == CENTER)
  {
    x = x - w / 2;
  }
  display.setCursor(x, y);
  display.print(text);
}
  return;

} // end drawString
```

**step 6:** 
change the city name in config.cpp, with your Chinese city name:

`const String CITY_STRING = "上海";`

step 7:
 compile and upload it. you got your Chinese city name.

You also can change other font name to support other language, like Japanese or Korea.

The code is not good and neat, but it works.