lmarzen/esp32-weather-epd #21: Include google Calendar

aggr0lite, 2023-03-10:
Hello again! 
I was trying to include google calendar items and making it configurable whether to show upcoming events or the outlook graph on the right. 

I built a node-red endpoint which is pulling events from iCal private links and responds with an array of days and events for each day in this format:

`{
    "days": [
        {
            "date": "Samstag, 11. März 2023",
            "events": [
                {
                    "summary": "awgaf",
                    "eventStart": "2023-03-11T08:30:00.000Z",
                    "calendarName": "CalendarA"
                }
            ]
        },
        {
            "date": "Samstag, 18. März 2023",
            "events": [
                {
                    "summary": "asgsdsaf",
                    "eventStart": "2023-03-17T23:00:00.000Z",
                    "calendarName": "CalendarB"
                },
                {
                    "summary": "aergherhae",
                    "eventStart": "2023-03-18T09:00:00.000Z",
                    "calendarName": "CalendarA"
                }
            ]
        }
    ]
}`

I was trying to implement it in your project but struggle how to pass the received response to a drawCalendarItems function in the rendere.cpp file. 

I must admit I'm not an expert in coding and this is really just a trial and error approach replicating what you did for parsing the OWM calls and drawing the content. 
Would you be willing to support this?

lmarzen, 2023-03-10:
I am not interested in supporting this feature, but if you create a fork with your changes, I check out your fork and see if I can help you resolve your issues. NOTE that if you create a fork, be sure not to sanitize your config.cpp.

aggr0lite, 2023-03-10:
It's been a while I used git, but I think I made it: [https://github.com/aggr0lite/esp32-weather-epd](url)
What do you need to know? 

Actually I'm trying to just pass the json object containing the array with all the event objects to the drawCalendarItems() function I defined in the renderer.cpp. I think I can figure out how to draw string and lines etc. but struggle to get the object array passed to the function.

lmarzen, 2023-03-10:
I am a little unsure what your problem is. If you add the changes you made to your fork, I can more easily see where you are at and where you are getting stuck.

lmarzen, 2023-03-11:
Ahh, you are running into an issue that plagued me at one point. Not enough ram. Specifically in the .dram0.bss segment which is where `static calendar_events_t calendar_api;` will be allocated.

Here are a few things you can do to reduce memory use in this segment. I suggest the following, you can use either one of these or a combination to fix your issue.
1.  Reduce the size of calendar_events_t. I see that you have CALENDAR_DAYS set to 99 which results in a huge array. Consider reducing this to something more reasonable if possible. If you make no other changes and reduce CALENDAR_DAYS to 31 the code will compile.
2. Reduce the page size for the e-paper display (effectively rendering only portions of the screen at a time). This is already done for the 3-color panel. In renderer.h and renderer.cpp increase the denominator of `GxEPD2_3C<GxEPD2_750c_Z08, GxEPD2_750c_Z08::HEIGHT / 2> display` from 2 to 4 or 8. (if you don't make any other changes you will need to change this to 8)

aggr0lite, 2023-03-12:
Very good feedback! I was reducing CALENDAR_DAYS as it cannot show much on the display anyway. Now I can compile and upload. 
However, stuck on calling the API now. It works via Postman.
But when I call it on the ESP it only returns errors.

When I use this, only the blank IP of the server as the host, I get the error below:

`    const char host[] = "xxx.xxx.xxx.xxx";
    const uint16_t apiPort = 1880;
    const char apiPath[] = "/xxxx";
    
    http.begin(client, host, apiPort, apiPath);`

Error:

`[  2968][E][WiFiClient.cpp:242] connect(): connect on fd 48, errno: 118, "Host is unreachable"
Calendar API error: -1 connection refused`

When using the host with "http://" it looks like this:

`    const char host[] = "http://xxx.xxx.xxx.xxx";
    const uint16_t apiPort = 1880;
    const char apiPath[] = "/xxxx";
    
    http.begin(client, host, apiPort, apiPath);`

Error:
`[ 11075][E][WiFiGeneric.cpp:1476] hostByName(): DNS Failed for http://xxx.xxx.xxx.xxx
Calendar API error: -1 connection refused`

Maybe you got any idea how to overcome this error?

closed 2023-03-15 by lmarzen

Marckau, 2023-10-17:
@aggr0lite Did you get it to work? Can you send a picture of it on the display?
Thanks