lmarzen/esp32-weather-epd #22: Error compile VS Studio

ParsonsGrahamJ, 2023-03-21:
Great project I am awaiting FireBeetle32 on order from China. I decided to configure and compile ready to upload once I get the board.

Getting an error from VS Code with one of the parameters.

src/main.cpp: In function 'void beginDeepSleep(long unsigned int&, tm*)':
src/main.cpp:105:58: error: call of overloaded 'String(uint64_t&)' is ambiguous
   Serial.println("Deep-sleep for " + String(sleepDuration) + "s");

VS Code fags these String(sleepDuration)

Any idea on fix ?

Thank you again for this great project, going to look at integrating outside ESP32 with temp/humd/Pressure sensor I already have over ESP-now. See if its possible.

Regards,

Graham

lmarzen, 2023-03-21:
Not an issue for me, but I just updated the project with a small change that should fix this error for you. Let me know if it works now.

ParsonsGrahamJ, 2023-03-21:
Wow that was a quick response Luke, I am impressed.

I did search for the reference to sleep Duration and suspected an issue with declaration. Unfortunately it still wont compile same error line.

Here is the full list output from VS Code

src/main.cpp: In function 'void beginDeepSleep(long unsigned int&, tm*)':
src/main.cpp:105:58: error: call of overloaded 'String(long long unsigned int&)' is ambiguous
   Serial.println("Deep-sleep for " + String(sleepDuration) + "s");
                                                          ^
In file included from C:/Users/Graham/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:159,
                 from src/main.cpp:18:
C:/Users/Graham/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:75:18: note: candidate: 'String::String(double, unsigned int)'
         explicit String(double, unsigned int decimalPlaces = 2);
                  ^~~~~~
C:/Users/Graham/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:74:18: note: candidate: 'String::String(float, unsigned int)'
         explicit String(float, unsigned int decimalPlaces = 2);
                  ^~~~~~
C:/Users/Graham/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:73:18: note: candidate: 'String::String(long unsigned int, unsigned char)'
         explicit String(unsigned long, unsigned char base = 10);
                  ^~~~~~
C:/Users/Graham/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:72:18: note: candidate: 'String::String(long int, unsigned char)'
         explicit String(long, unsigned char base = 10);
                  ^~~~~~
C:/Users/Graham/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:71:18: note: candidate: 'String::String(unsigned int, unsigned char)'
         explicit String(unsigned int, unsigned char base = 10);
                  ^~~~~~
C:/Users/Graham/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:70:18: note: candidate: 'String::String(int, unsigned char)'
         explicit String(int, unsigned char base = 10);
                  ^~~~~~
C:/Users/Graham/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:69:18: note: candidate: 'String::String(unsigned char, unsigned char)'
         explicit String(unsigned char, unsigned char base = 10);
                  ^~~~~~
C:/Users/Graham/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:68:18: note: candidate: 'String::String(char)'         explicit String(char c);
                  ^~~~~~
C:/Users/Graham/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:66:9: note: candidate: 'String::String(StringSumHelper&&)'
         String(StringSumHelper &&rval);
         ^~~~~~
C:/Users/Graham/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:65:9: note: candidate: 'String::String(String&&)'
         String(String &&rval);
         ^~~~~~
C:/Users/Graham/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:62:9: note: candidate: 'String::String(const String&)'
         String(const String &str);
         ^~~~~~
*** [.pio\build\firebeetle32\src\main.cpp.o] Error 1

Graham

lmarzen, 2023-03-21:
Hi Graham. This is interesting.

It looks like you don't have a definition in your WString.h for unsigned long long. This was added in February of this year: https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/WString.h

Can you tell me what versions you see for framework-arduinoespressif32 when you build?

It should say something like this...

> PACKAGES: 
>   \- framework-arduinoespressif32 @ 3.20005.220925 (2.0.5)

ParsonsGrahamJ, 2023-03-21:
Me again,

I had a look at the library WString.h and could not see a reference for a "unsigned long long" I did see a reference for "unsigned long". 

Just changed it to "unsigned long sleepDuration = 0;" and it worked. Now compiles completely.

Will keep you posted once I complete the build. Will see if I can intergrate a remote sensor for real time outside temp/humd/pressure.

ParsonsGrahamJ, 2023-03-21:
Here is a copy of the WString references I have

        explicit String(char c);
        explicit String(unsigned char, unsigned char base = 10);
        explicit String(int, unsigned char base = 10);
        explicit String(unsigned int, unsigned char base = 10);
        explicit String(long, unsigned char base = 10);
        explicit String(unsigned long, unsigned char base = 10);
        explicit String(float, unsigned int decimalPlaces = 2);
        explicit String(double, unsigned int decimalPlaces = 2);
        ~String(void);

lmarzen, 2023-03-21:
Yes, looks like you may be on an older version of framework-arduinoespressif32. I'm not sure why platformio is not automatically getting the latest version.

ParsonsGrahamJ, 2023-03-21:
VS Code is still very buggy and a real pain when building code. Had lots of issues with some WLED projects and libraries not downloading the latest. The guys that write the code for WLED have also had issues with some people not getting the current updates, its hit and miss.

framework-arduinoespressif32 @ 3.20003.220613 (2.0.3)

Just clicked the update and it said is up to date, so an issue with VS Code, hope that helps but it worked with the above and compiled, so hope it works once I get all the hardware together.

ParsonsGrahamJ, 2023-03-21:
Thank you so much for the help looks the the issue was indeed the PlatformIO not updating. 

Looked at there forum PlatformIO and looks like others have had issues. Have manual updated both ESP32 and ESP8266 as they were out of date and compiled with your original code no issues.

Learn something new every day, here is the manual update method,

**PlatformIO will install the latest Arduino Core for ESP32 from https://github.com/espressif/arduino-esp32. The [Git](https://git-scm.com/) should be installed in a system. To update Arduino Core to the latest revision, please open [PlatformIO IDE](https://docs.platformio.org/en/latest/integration/ide/pioide.html#pioide) and navigate to PlatformIO Home > Platforms > Updates.**

It updated all the outdated version with the above.

Will keep you posted once complete hardware

Thanks again Graham

lmarzen, 2023-03-21:
Thank you for sharing the solution. Good luck with your project and integrating esp-now, sounds like an exciting project.

lmarzen, 2023-03-21:
I discovered that I can specify the version of the espressif framework to use in the platformio.ini file. This should prevent issues like this in the future. I will add this. Thanks again Graham.

lmarzen, 2023-03-21:
e2979d7

closed 2023-03-21 by lmarzen

ParsonsGrahamJ, 2023-03-21:
Yes on the same link from the PlatformIO it does say you can force versioning within the PlatformIO.ini file. It is still annoying that this is not done with the usual update method on PlatformIO. Been court out a few times with version on PlatformIO. When I updated both the ESP32 and ESP8266 I had lots of other programs started behaving correctly too on compile. Completely forgot about checking if it was updated. Learn something new.

lmarzen, 2023-03-21:
I agree. I would have expected that PlatformIO would have automatically checked for updates for project dependencies.