lmarzen/esp32-weather-epd #23: Outdoor Temp Sensor

ardiross, 2023-03-22:
I just built this project yesterday and love it!!  Have you considered maybe adding a wireless temp sensor to the project?

lmarzen, 2023-03-22:
Glad to hear that!

I have thought about adding a wireless temperature sensor (or some capability to get current weather info from a weather station outside) but have not made any real effort to make that happen.

nglessner, 2023-03-22:
I'm attempting to get temp (and humidity) values from home assistant via
their rest API.  There shouldn't be any additional libraries involved.
Simply make an http get call to the API and use those values. That being
said, in my limited effort, I haven't gotten past a 401 response, so I'm
not sure what's going on with that.  MQTT would be another good option if
you're already using it.

ayh20, 2023-03-22:
I'm planning to use HACS as well as. It's a call to the state interface
http://192.168.1.40:8123/api/states/sensor.ShedTemp
You need to create an API token and pass that as a bearer token but it works nicely

nglessner, 2023-03-22:
Yep, I have all of that. I'm doing an http.GET() and adding the
Authorization header. (note the HA_APIKEY starts with "Bearer __TOKEN___")
I've tested it with curl and it works fine. I might have to bust out
wireshark to see what it's actually firing
off.
    HTTPClient http;
    http.addHeader("Authorization", HA_APIKEY);
    //http.addHeader("Content-Type", "application/json");
    http.begin(client, HA_ENDPOINT, HA_PORT, uri);
    httpResponse = http.GET();

ayh20, 2023-03-22:
FWIW ... The HACS UI for getting the token is horrible ..... the token is 148 bytes long ... the UI makes it really hard to get the full value and i had 401's because i'd not copied the complete key

ayh20, 2023-03-22:
Apologies i didn't spot that in your earlier response. That does sound right ... header name  of "Authorization" with "Bearer ey.....xxx" is what i would expect.   Some sort of string truncation error ?
Sadly i don't have all the bits to test this myself as the FireBetle was DOA ... should be able to play later in the week.

nglessner, 2023-03-22:
The issue was that you can't add an Authorization header manually. You must
use their built in method.

    http.addHeader("Authorization", HA_APIKEY); // does not work

    http.setAuthorization("HA_APIKEY"); // works
    http.setAuthorizationType("Bearer");

ParsonsGrahamJ, 2023-03-22:
I was messaging Luke about this yesterday. I have an ESP32 with Temp/Humd/Pressure for local outside conditions. 

I am awaiting parts to build this project first but was thinking of looking down the lines of esp-now messaging real time data to the this project to give real local values. I am aware that esp-now and wifi internet are not easy to coexist on ESP32 but possible.

Another option is to use data from my Netatmo weather station which is currently linked to my HA setup. This weather station has has Temp/Humid/Pressure/Wind Speed and Rainfall so a lot more local value to upload.

Will be watching what comes up here for inspiration.

nglessner, 2023-03-23:
@ParsonsGrahamJ I've thought of this as well.  I have a network of ESPNow devices outdoors that are all battery powered. That would be nice to get a local outdoors temperature, but there's a problem.  My sensors spend the vast majority of their lives in deep sleep to conserve battery. The weather-epd is also mostly in deep-sleep and battery powered. How do you solve the problem of making sure they're awake at the same time? or do you provide mains power to one of the 2 devices?  My workaround would be to have a mains powered server (such as Home Assistant) to store the outdoor (and indoor) sensor data. Then you can use the weather-epd to poll those when it's awake.

ayh20, 2023-03-24:
@nglessner This is exactly what i have... sensors around the home and garden that log data to HA and then i can read it from there. It's a std design... IoT devices feeding data to a Edge or core server, apps that need the data get it from the edge/core server

ParsonsGrahamJ, 2023-03-24:
> @nglessner 

Yes I did consider the problem of sleep and wakeup of sensors and main weather station. The data collected would need to be stored until the main weather station wakeup and then transmitted. 

Have a look at this instructable someone has created to do this very thing. 
https://www.instructables.com/An-ESP-Now-Home-Weather-Station/

I was going to try this as an option before using my HA setup so I could keep them separate but might just use my existing HA setup with all its sensor data and pole the data when the weather station wakes up.

Once I get all the hardware together and make this will see what works the best.