lmarzen/esp32-weather-epd #25: Localisation: Wind speed

jrial, 2023-03-22:
I noticed that wind speed for Metric uses the SI unit of m/s. However, that's not how we measure speeds over here in anything other than a scientific setting, and very specific scenarios like muzzle speeds of firearms. Colloquially, we measure in km/h. With a conversion factor of * 3.6 from m/s to km/h. But some sites also report in "[Beaufort](https://en.wikipedia.org/wiki/Beaufort_scale)", which has a slightly more complex conversion formula.

Air pressure has a similar issue: the current "Metric" defaults to hPa, but mbar is also used (moreso in the past). Fortunately, both units are interchangeable: 1 atmosphere equals 1013hPa, which equals 1013mbar. So only the label needs to be swapped, for those that prefer to read "mbar".

I'm willing to create a pull request for this, but I was wondering which approach you'd prefer:

1) Just update m/s to km/h and apply the conversion factor, disregard Beaufort and mbar.
2) create multiple "metric variants", e.g. `UNITS_METRIC_SI`, `UNITS_METRIC_KPH`, `UNITS_METRIC_BEAUFORT`, and a setting to toggle between hPa and mbar.
3) Split the unit system out into separate systems for temperature, wind speed and air pressure, so people can configure these independently. Imperial/Metric would then only apply to temperature, and renamed to `UNITS_TEMPERATURE_IMPERIAL`. New config keys would be added for `UNITS_WINDSPEED_KPH` (and `*_MPS` or `*_SI` and `*_MPH`), `UNITS_PRESSURE_HPA` and `UNITS_PRESSURE_MBAR` (and whatever is currently defined for imperial; need to look that up).

lmarzen, 2023-03-22:
Thank you for bringing this to my attention (especially the m/s to km/h). I like solutions 1 and 3. You are welcome to implement this yourself and create a pull request.

jrial, 2023-03-22:
Great! I'll look into it then.

lmarzen, 2023-03-26:
I found some time today and added new options for units. bd0a8c8e9957f4f25f3c31418776102691f65d99 8dd6a8a7856c8a02d64b4f0d6036f9bec1e4cde4

jrial, 2023-03-27:
I''m looking at the changes. Any reason you use a ` instead of ° in the temperature labels?

Small issue: Kelvin is just Kelvin, not degrees Kelvin. So its label is just K with nothing in front of it.

ayh20, 2023-03-27:
> I''m looking at the changes. Any reason you use a ` instead of ° in the temperature labels? I see they're still rendered as ° on the display though, how does that work?
@jrial It's because of the custom fonts being used ... the code point for ` is actually displayed as the °

lmarzen, 2023-03-27:
@jrial Thank you for correcting my error of using the degree symbol in the Kelvin units. (Fixed now)

Regarding ` instead of °. @ayh20 is right. 
For more context, when I first created this project, it only supported the 7-bit ASCII character set. The 7-bit ASCII character set does not include the ° symbol, so I created a custom font to map ` to °. Since then, I received requests to expand the character set. I still have some hopes to add UTF-8 support (just the Basic Multilingual Plane), but this has a number of technical challenges associated with it. So to satisfy the requests of people using this project, I expanded the character set to the 8-bit extended ASCII set. The extended character set does have the degree symbol in it, so I am considering removing the custom mapping in my font because it is redundant now.

closed 2023-03-29 by lmarzen