lmarzen/esp32-weather-epd #26: Compile error " no such file pins_arduino.h"

coldfire100, 2023-03-22:
Hi Luke
First off, great project nice work! One problem I can't seem to get around is the error 
"fatal error: pins_arduino.h: No such file or director"
I cloned the project from Github and have the latest files and platformio.ini. Any suggestions on where to look for this.

I can post the full output from the build if needed.

I'm pretty new at VsCode and PlatformIO so any help would be much appreciated.

Thanks

Mark

lmarzen, 2023-03-23:
Hi Mark,

Yeah go ahead and share your full output. Additionally, can you confirm what esp32 board you are using?

-Luke

coldfire100, 2023-03-23:
Hi Luke
Sorry for the late response. The board the is DF Robot Firebeetle2. I have 2 on order. This is a clone of your project. I didn't change anything in the settings. Below is the output from the build.

Executing task: C:\Users\ZZZ\.platformio\penv\Scripts\platformio.exe run 

Processing dfrobot_firebeetle2_esp32e (board: dfrobot_firebeetle2_esp32e; platform: espressif32 @ 6.1.0; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/dfrobot_firebeetle2_esp32e.html
PLATFORM: Espressif 32 (6.1.0) > DFRobot Firebeetle 2 ESP32-E
HARDWARE: ESP32 80MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20007.0 (2.0.7)
 - tool-esptoolpy @ 1.40500.0 (4.5.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 40 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Adafruit BME280 Library @ 2.2.2
|   |-- Adafruit BusIO @ 1.14.1
|   |   |-- Wire @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|   |-- SPI @ 2.0.0
|   |-- Adafruit Unified Sensor @ 1.1.9
|-- Adafruit BusIO @ 1.14.1
|   |-- Wire @ 2.0.0
|   |-- SPI @ 2.0.0
|-- Adafruit Unified Sensor @ 1.1.9
|-- ArduinoJson @ 6.21.0
|-- GxEPD2 @ 1.5.0
|   |-- Adafruit GFX Library @ 1.11.5
|   |   |-- Adafruit BusIO @ 1.14.1
|   |   |   |-- Wire @ 2.0.0
|   |   |   |-- SPI @ 2.0.0
|   |   |-- Wire @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- Adafruit BusIO @ 1.14.1
|   |   |-- Wire @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|   |-- SPI @ 2.0.0
|-- pollutant-concentration-to-aqi
|-- HTTPClient @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- WiFiClientSecure @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|-- SPI @ 2.0.0
|-- WiFi @ 2.0.0
|-- Preferences @ 2.0.0
|-- Wire @ 2.0.0
Building in release mode
Compiling .pio\build\dfrobot_firebeetle2_esp32e\src\_strftime.cpp.o
Compiling .pio\build\dfrobot_firebeetle2_esp32e\src\api_response.cpp.o
In file included from C:/Users/ZZZ/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:79,
                 from C:/Users/ZZZ/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,
                 from include/_locale.h:22,
                 from src/_strftime.cpp:29:
C:/Users/ZZZ/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.h:29:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

[... 86 lines cut ...]
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

 #include "pins_arduino.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\dfrobot_firebeetle2_esp32e\src\locales\locale_en_gb.cpp.o] Error 1
========================================================================= [FAILED] Took 260.67 seconds =========================================================================

Environment                 Status    Duration
--------------------------  --------  ------------
dfrobot_firebeetle2_esp32e  FAILED    00:04:20.669
===================================================================== 1 failed, 0 succeeded in 00:04:20.669 =====================================================================
 *  The terminal process "C:\Users\ZZZ\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

I have your latest platformio.ini file. Not sure if anything else needs to be updated.
If you could take a look at this.

Thanks

Mark

lmarzen, 2023-03-26:
hmmm.. can you see if there are any updates for arduino core or espressif, etc? 
> Navigate to PlatformIO Home > Platforms > Updates.

coldfire100, 2023-03-26:
Hi Luke
I did a bit of research and found a reference to the "arduino.pins.h" error. The suggestions was to delete the folders in the 
user/.platformio/packages that referenced the esp32.

I wound up deleting the folders:

framework-arduinoespressif32@3.10004.210126
tool-esptoolpy
framework-arduinoespressif32
toolchain-xtensa-esp32@8.4.0+2021r2-patch5

I did a clean and rebuild. Lo and behold it compiled without the errors. Any idea why this would have worked?

Anyway,  I don't have the boards in yet to test it. They should be here in a couple of weeks.
Hopefully all will work.

Thanks for looking into this. 

Mark

lmarzen, 2023-03-26:
No idea. Sounds like platformio being buggy, smh. Glad to hear you were able to resolve your issue.

closed 2023-03-26 by lmarzen