lmarzen/esp32-weather-epd #28: Excessive update requests tied to: Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled

ChiefPoints, 2023-04-01:
I periodically had an issue where the project would burn through all 1,000 daily requests with openweathermap's service in short order, resulting in a "too many requests" error. I hooked up the serial monitor and let it run for over a week and it finally error'd out again. The errors with core dumps begin on like 6372 of the logfile, towards the end. I've left the preceding entries in the log so you can see normal operation. I intercepted the error before it burned through all my daily requests, looks like it was retrying every few seconds. Let me know if you need anything else.

[putty-esp32 - core dumps.txt](https://github.com/lmarzen/esp32-weather-epd/files/11128013/putty-esp32.-.core.dumps.txt)

lmarzen, 2023-04-01:
> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled`

The error message you received indicates that there was an attempt to load data from an invalid memory address, which triggered a reset. This error can occur for a number of reasons, such as using an invalid pointer or accessing memory that has not been properly initialized.

To troubleshoot the issue, we need to isolate the code that is causing the error. The bug may be occuring when receiving the API responses. This is a nasty bug because it occurs so infrequently, which leads me to suspect that some API responses are not being handled correctly. (maybe the response is too big in some cases? i am unsure).

If this is because of an API response, then it would be helpful to get a copy of the .json. If you are unsure how to do this, the easiest way would be to type this into your browser when the error is happening... 
`api.openweathermap.org/data/3.0/onecall?lat={lat}&lon={lon}&lang=en&units=standard&exclude=minutely&appid={API key}`
(replacing {lat} {lat} and {API key} with the values that you are using for your weather display project)
Then you can `Ctrl + s` and save the response as a .json.

I have added some more serial print messages that should give us more confidence about where this error is occurring.

Thank you for putting in some serious work to help debug this. The serial log is excellent work.

ChiefPoints, 2023-04-01:
You are welcome. I'll update my device with the new code and see if we can zero in on this. I'll try to get this done tonight and it usually takes up to 10 days to see the error, I'll update with new logs at that time.

lmarzen, 2023-04-01:
Sounds good. Good luck, or rather bad luck?

ChiefPoints, 2023-04-07:
The error happened again, but I was out for a couple of days and wasn't able to try to get a .json download. The crash happens after it reads the BME280, and it resolved itself after about 500 crashes. Here's the log file:
[putty-esp32-crash after code update.log](https://github.com/lmarzen/esp32-weather-epd/files/11181918/putty-esp32-crash.after.code.update.log)

lmarzen, 2023-04-07:
Hi @ChiefPoints 

This narrow's it done some. I want to look more into this later this weekend. Could you provide a copy of your config.cpp and config.h, so that I can eliminate some variables? Please be sure to remove any sensitive information from your config.cpp, like wifi credentials, api key, etc.

Thanks,
-Luke

lmarzen, 2023-04-07:
From: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html#loadprohibited-storeprohibited 

> LoadProhibited, StoreProhibited
These CPU exceptions happen when an application attempts to read from or write to an invalid memory location. The address which has been written/read is found in the EXCVADDR register in the register dump. If this address is zero, it usually means that the application has attempted to dereference a NULL pointer. If this address is close to zero, it usually means that the application has attempted to access a member of a structure, but the pointer to the structure is NULL. If this address is something else (garbage value, not in 0x3fxxxxxx - 0x6xxxxxxx range), it likely means that the pointer used to access the data is either not initialized or has been corrupted.

In your case the EXCVADDR register indicates 0xffffffff. Now we know it is probably a pointer that in some rare cases is isn't getting initialized. This will take some more investigating to resolve.

lmarzen, 2023-04-08:
If you can decode your stack trace, we would know exactly were the error is occuring. `0x401630c2:0x3ffb1ff0 0x400db6d2:0x3ffb2030 0x400d9bde:0x3ffb20d0 0x400e7c72:0x3ffb2290`

To do this, you need the `firmware.elf` file that you were running at the time of this error and you need to find where your platformio packages are, specifically, you need to find the tool called `xtensa-esp32-elf-addr2line`.

(I'm on linux and found mine here... $HOME/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-addr2line)

Then you can run the following command...

`/home/luke/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-addr2line -pfiaC -e /home/luke/projects/esp32-weather-epd/platformio/.pio/build/dfrobot_firebeetle2_esp32e/firmware.elf 0x401630c2:0x3ffb1ff0 0x400db6d2:0x3ffb2030 0x400d9bde:0x3ffb20d0 0x400e7c72:0x3ffb2290`

Of course, replacing the paths to correspond to the locations on your machine.

ChiefPoints, 2023-04-08:
I check it out as soon as I can, may not be until tomorrow. Just setup a new NAS and I'm knee deep in moving data.

ChiefPoints, 2023-04-08:
I've attached the output, sanitized the home folder name with XXXXX, let me know if you need anything else.

[addr2line_output.txt](https://github.com/lmarzen/esp32-weather-epd/files/11182384/addr2line_output.txt)

lmarzen, 2023-04-08:
Good news, thanks to your decoded trace you provided I managed to recreate the issue. It is an out of bounds array access caused when the wind degree is between 341 and 348 degrees. Should be a quick fix. Shows that I should have tested this code better.

lmarzen, 2023-04-08:
I had used integer division instead of floating point division. smh. Hopefully, your issue is no more.

I am marking this issue as complete. If your problem persists, leave a comment, and I will reopen this issue.

closed 2023-04-08 by lmarzen

ChiefPoints, 2023-04-08:
Outstanding, thanks!