lmarzen/esp32-weather-epd #31: Description for european air quality index is upside down?

kaerbr, 2023-04-15:
I just installed everything. But I came across an issue(?) with the description of the air quality if the european air quality index is selected.

Today the air quality index is `14` (wich should be a really good value (lower = better)).
For the en_US translation this value is described as `Very Low`.
As said in https://en.wikipedia.org/wiki/Air_quality_index#Europe (second paragraph of "CAQI"): `The CAQI is a number on a scale from 1 to 100, where a low value means good air quality and a high value means bad air quality.`
So in my understanding the description should be `Very High` instead.

The issue might come from the [table below](https://en.wikipedia.org/wiki/Air_quality_index#Europe), which lists the `Qualitative name` and the corresponding `Index or sub-index`. In my understanding the `Qualitative name` answers the question: "How high or low is the **pollution**?" but I think the question that should be answered for the `air quality (index)` is: "How good or bad ist the **air** (quality)?"

So in conclusion the char array for the air quality with european calculation should not be from low to high
```cpp
const char *EUROPE_CAQI_TXT[5] = 
{
  "Very Low",
  "Low",
  "Medium",
  "High",
  "Very High",
};
```
but the other way round: from high to low (as the array for the us air quality is):
```cpp
const char *EUROPE_CAQI_TXT[5] = 
{
  "Very High",
  "High",
  "Medium",
  "Low",
  "Very Low"
};
```

I hope this is understandable. 🙈 
If you think the same I would create a merge request. (:

ps. the same principal applies to `CANADA_AQHI_TXT`, `HONG_KONG_AQHI_TXT`, `UNITED_KINGDOM_DAQI_TXT`.

lmarzen, 2023-04-15:
Hello @kaerbr,

I think I understand your concern. 

I'll quickly explain why it is the way it is. Feel free to let me know if you still disagree. 

I used the Qualitative Name/Descriptor/Category outlined in the official standard for each AQI. Some AQI's attempt to fulfil different purposes, some present amount of air pollution like CAQI, and others like the US AQI present air quality closer to the sense that you are suggesting.

For the EU's CAQI...

from [http://airqualitynow.eu/about_indices_definition.php](https://web.archive.org/web/20220324185850/http://airqualitynow.eu/about_indices_definition.php)
and
from [http://www.airqualitynow.eu/download/CITEAIR-Comparing_Urban_Air_Quality_across_Borders.pdf](https://web.archive.org/web/20220303184040/http://www.airqualitynow.eu/download/CITEAIR-Comparing_Urban_Air_Quality_across_Borders.pdf)
> These indices have five levels using a scale from ‘0‘ (very low) to ‘> 100’ (very high). It presents a relative measure of the amount of air pollution.

For the US's AQI...

from [https://www.airnow.gov/sites/default/files/2020-05/aqi-technical-assistance-document-sept2018.pdf](https://www.airnow.gov/sites/default/files/2020-05/aqi-technical-assistance-document-sept2018.pdf)

> The purpose of the AQI is to inform people about their air quality so they can take steps to protect their health.

kaerbr, 2023-04-15:
Ah okay, I see. So technically this is correct.

For me personally thats not intuitive. But the solution is to use the US's AQI with german translations. (: (I'm from Germany).

closed 2023-04-18 by lmarzen