lmarzen/esp32-weather-epd #34: ePaper Screen does not work

em-steven, 2023-05-08:
Hello! Thank you for the nice project!

This is a bit outside the scope of this project, but I thought I would try anyway. 

I bought [this screen](https://www.waveshare.com/7.5inch-e-paper.htm) and this [ESP32 board](https://www.waveshare.com/e-paper-esp32-driver-board.htm) for the screen. This ESP board has different pins for SPI, [found here](https://www.waveshare.com/wiki/E-Paper_ESP32_Driver_Board#Pin). Previously the e-paper screen was working with them. But when I update config.cpp with the same pins, the screen doesn't react. 

I can see via the serial monitor that the code is working because it's connecting to my Wifi and making an API request to Openweather. 

Any ideas what It could be?

lmarzen, 2023-05-08:
I do not have any experience with that particular board. Can you clarify what you mean by "Previously the e-paper screen was working with them. But when I update config.cpp with the same pins, the screen doesn't react."?

em-steven, 2023-05-08:
Well upon further inspection, the ribbon cable going into the screen is torn and several pins are not connected. So that solves that mystery. 

Thanks for a quick reply. 👍

closed 2023-05-08 by em-steven

lmarzen, 2023-05-08:
Sorry to hear that, but glad you got it figured out:)

bob-u, 2023-10-06:
> Well upon further inspection, the ribbon cable going into the screen is torn and several pins are not connected. So that solves that mystery.
> 
> Thanks for a quick reply. 👍

Can you share your settings?  I have the same board, it works and I cannot get it to work.  My settings:
const uint8_t PIN_EPD_BUSY = 25; //13; // 5 for micro-usb firebeetle
const uint8_t PIN_EPD_CS   = 15; //2;
const uint8_t PIN_EPD_RST  = 26; //21;
const uint8_t PIN_EPD_DC   = 27; //22;
const uint8_t PIN_EPD_SCK  = 13; //18;
const uint8_t PIN_EPD_MISO = 18; // 19 Master-In Slave-Out not used, as no data from display
const uint8_t PIN_EPD_MOSI = 14; //23;

YuriStruszczynski, 2023-12-17:
> > Well upon further inspection, the ribbon cable going into the screen is torn and several pins are not connected. So that solves that mystery.
> > Thanks for a quick reply. 👍
> 
> @em-steven Can you share your settings? I have the same board, it works with waveshare demos but I cannot get it to work with this project. My settings: const uint8_t PIN_EPD_BUSY = 25; //13; // 5 for micro-usb firebeetle const uint8_t PIN_EPD_CS = 15; //2; const uint8_t PIN_EPD_RST = 26; //21; const uint8_t PIN_EPD_DC = 27; //22; const uint8_t PIN_EPD_SCK = 13; //18; const uint8_t PIN_EPD_MISO = 18; // 19 Master-In Slave-Out not used, as no data from display const uint8_t PIN_EPD_MOSI = 14; //23;

Same here. Did you find a working solution? :-)