lmarzen/esp32-weather-epd #36: What does rxOWM[2] identifier stand for?

AimPizza, 2023-06-23:
I am currently trying to understand the code (still learning lol) in order to get a different API working instead of onecall.
So I cam across said `int rxOMW[2] ` which holds http response codes. But I couldn't find any naming convention or something that explains the 'rx' part.
Great project by the way, love it!

lmarzen, 2023-06-23:
Thanks @AimPizza.

rx means received. Maybe not the most fitting name anymore. When I first wrote that code I think it stored a boolean, but at some point I repurposed it without properly renaming it.

-Luke

lmarzen, 2023-06-23:
If you have any suggestions that would improve clarity, feel free to create a pull request.

lmarzen, 2023-10-20:
Renamed.

closed 2023-10-20 by lmarzen