lmarzen/esp32-weather-epd #37: WiFi Manager Integration

jfuji6, 2023-07-08:
Hello, 
Are you interested in a PR to integrate [WiFi Manager](https://github.com/tzapu/WiFiManager)?  I have local branch with it integrated so, I can change WiFi networks/passwords without having deploy a new build.  

WiFi Manager also supports custom fields, so I've considered adding custom fields to also set.  However I haven't gone down that rabbit hole yet.  
- OWM_APIKEY (Not sure if there are any security implications.  With it complied it would be harder to get to.  If I were to add it to configs, it would likely just be saved in json format using SPIFFS).
- CITY_STRING (In case I move)
- LAT (CITY_STRING would be useless without this).
- LON (CITY_STRING would be useless without this).
- TIMEZONE (In case I move)

lmarzen, 2023-07-08:
Hi @jfujitani,

This would be an interesting integration. Many people would appreciate this feature. Admittedly, I am not very familiar with this library and only just now skimmed some of the documentation. You are welcome to create a PR and we can work to get it merged into the project. I like the idea of being able to change more commonly used options.

jfujitani, 2023-07-08:
Hi, @lmarzen,

Thanks for the reply.  I'll create a PR for the initial WiFi only integration.
There's some things you'll want to look at before going forward with it.  I'll have it up in a little bit.

jfujitani, 2026-10-24:
@lmarzen *I'm back* [took another swing at this](https://github.com/lmarzen/esp32-weather-epd/pull/256).  Let me know if this approach is better for you.  No worries if it's not in-line with where you want to take the project.   I'll continue to test and experiment on my end if not.  
- John