lmarzen/esp32-weather-epd #39: Is it possible to have the wind direction shown in degrees?

brookenmire, 2023-07-09:
Hi,
Just a feature request/enhancememnt.
Is it possible to have the wind diricetion shown in degrees rather than an arrow Icon.
i.e. 270 @ <wind speed>
Had a look at the code but its beyond my understanding.

Built the Station, work great, thanks!

lmarzen, 2023-07-12:
Seems like a simple enough request. I can add it once I get more time, I have been pretty busy lately.

brookenmire, 2023-07-13:
Thanks Luke, much appreciated.

lmarzen, 2023-12-17:
This has been in the backlog for too long. Sorry about that but it is finally done. You can select from the Arrow, a Number like you suggested or Compass Point Notation (i.e. NE, SW, etc). You can even use the arrow in combination with the Number or CPN.

closed 2023-12-17 by lmarzen

brookenmire, 2023-12-21:
Brillant, thanks Luke.
I will recompile and install the new firmware.
Display has been a centre piece on my counter and used often throughout the
day.
Love the e-ink displays and your layout.
Thanks again.