lmarzen/esp32-weather-epd #40: Support for Waveshare 7.5 e ink v1

JaquelineP, 2023-09-05:
Hi @lmarzen 
I like to use your project. However, I only have the old e ink display with a resolution of 640 × 384px.
Can you add support for this display as well. I found the config in 'config.h' line 24f but I don't know what needs to be changed for the smaller resolution.
Thanks in advance.

lmarzen, 2023-09-05:
Hi @JaquelineP,

Thanks for the request. It would be a somewhat tedious undertaking to create a version of the software for an alternative resolution. There currently does not exist any scaling logic so someone would need to either manually adjust the size and position of every icon and piece of text or a not insignificant development effort would be needed to add scaling logic. I am not against this idea, but I simply do not have the time nor the Waveshare 7.5 e-ink v1 panel to be able to accomplish this. If you or another community member want to add support for this panel I would be happy to merge it into the main project.

lmarzen, 2023-09-10:
@JaquelineP If you want to make any revisions in the future for the v1 layout feel free to open a pull request. :)

closed 2023-09-10 by lmarzen