lmarzen/esp32-weather-epd #42: Waveshare e-Paper Driver HAT rev2.3

Spanholz, 2023-09-07:
I got a new 7.5 3-colour display from Waveshare this week including a Waveshare e-Paper Driver HAT rev2.3. The new HAT is not documented on Waveshares webpage yet and is different from the previous versions. It features 9 pins, with an additional PWR pin. The display does not start with VCC or PWR only connected. But with PWR connected to second 3.3V and therefore disconnecting the BME it works. I will update if the PWR pin can be put on same 3V3 as the VCC from the hat. 

[image]

Spanholz, 2023-09-07:
Putting PWR and VCC on the same 3V3 pin works. The other PIN can than be used by BME

lmarzen, 2023-09-09:
Thanks for investigating and documenting this.

I just found this page talking about the changes...
https://www.waveshare.net/wiki/E-Paper_Driver_HAT

Translating to english...

> Version changes:
> 
>     The RST pin now only controls reset and no longer controls power on and off.
>     Add a PWR pin to control power on and off
>     Replace the inductor with a smaller, thinner package
>     PH2.0 8Pin interface replaced with GH1.25 9Pin interface
>     The version number silk screen is updated to Rev2.3
> 
>     If you need to upgrade to this version from a previous version, you can choose one of the following methods for compatibility:
> 
>     Update the program to the latest (Arduino program only supports 7.5inch e-Paper, 7.5inch e-Paper (B), 5.83inch e-Paper, 5.83inch e-Paper (B), 2.9inch e-Paper (D), 2.13 inch e-Paper (D) six screens)
>     Connect PWR pin to 3.3V
>     Solder a 47K (or similar) resistor to the empty pad on the back

Appears you are right to connect PWR to 3.3V. I'll keep an eye on the epaper driver library this project uses and see if they provide any updates.

nonpayer, 2023-09-30:
I apologize, but I'm a beginner. I also received the Waveshare e-Paper Driver HAT rev2.3. I'm doing something wrong, because when I upload, I have 141 problems in renderer.app. Can anyone help me as a complete beginner? I followed the instructions, but I probably have something wrongly connected because I have HAT rev 2.3. and there is another in the manual...
[image]

[image]

lmarzen, 2023-09-30:
If you are getting a lot of problems in renderer.cpp, your config.h file may be misconfigured.

lmarzen, 2023-09-30:
Additionally can you share the exact model of panel you are using as well as the model of esp32 to make it easier to assist in troubleshooting.

nonpayer, 2023-09-30:
800×480, 7.5inch E-Ink display HAT for Raspberry Pi and e-Paper Driver HAT ver 2.3 with 9 pins, with an additional PWR pin.

FireBeetle 2 ESP32-E

Update is success without error, but i have only black edges on the display...

lmarzen, 2023-09-30:
Try reverting your config.h back to the default.

nonpayer, 2023-10-02:
Thanks for advice. I tried to use the default config.h and the upload was successful. But I have a 9-pin HAT and with an additional PWR pin. As Spanholz already wrote, the display only works with PWR connected to second 3.3V and therefore disconnecting the BME, or PWR and VCC on the same 3V3. But how do I plug PWR and 3V3 into the same connector. ?

lmarzen, 2023-10-02:
You will need to make a splitter. or solder both wires to the same 3v on the esp32

sonett12, 2023-10-03:
first of all thank you very much for your project @lmarzen I got myself the same hardware and have it running a few days now, works like a charm :-)

I also got the rev2.3 e-paper hat with this additional pwr pin to enable the display. I looked it up in the schematic here
https://files.waveshare.com/upload/8/8e/E-Paper_Driver_HAT.pdf 

[image]

For my understanding the additional pin can be controlled to activate/disable the display but is not powering the display, so I just used a GPIO pin for this, I connected the pwr pin to D6 pin 14 and added the activation in the main.ccp as such (lines 318..320 + 335):

[image]

for me that was enough, maybe others you can do so as well. Could be nice to have it an option in the project source code to configure this pwr pin as well.

nonpayer, 2023-10-03:
> You will need to make a splitter. or solder both wires to the same 3v on the esp32

I did it and everything works fine. Thanks for help.

lmarzen, 2023-10-03:
@sonett12 Thanks for investigating this further and sharing your results. I'll be adding an option for this and adding some documentation when I get a chance.

Marckau, 2023-10-13:
I tried the @sonett12 code and hooked the power pin to the D6 and updated the code as well as just soldering both the VCC and PWR on the 3V3 line. Same result the screen works and displays all the correct information for a brief moment then goes black. Any idea why? I have the black and red display. 
[image]
[image]

lmarzen, 2023-11-19:
I have suggested users purchase the DESPI-CO2 adaptor board instead of the waveshare hat going forward.

closed 2023-11-19 by lmarzen

nicoh88, 2024-06-06:
> the pwr pin to D6 pin 14

Hi @sonett12, you say you would use the **D6 / 14** for the new e-Paper HAT rev2.3 for the additional **PIN PWR**.

According to the [wiring diagram](https://github.com/lmarzen/esp32-weather-epd/blob/main/showcase/wiring_diagram_waveshare_rev22.png) from lmarzen, **D6 / 14** is already used for **BUSY**. **Do you then not use BUSY at all or how did you do it?**

I actually wanted to use **D9 / 2** for PWR, but according to the documentation of the[ FireBeetle Board ESP32-E (SKU DFR0654)](https://wiki.dfrobot.com/FireBeetle_Board_ESP32_E_SKU_DFR0654) D9 / 2 has a remark: "_For controlling onboard LED by outputting digital signal_"

Hence my question about what you did with BUSY.
Thank you in advance for your answer.