lmarzen/esp32-weather-epd #43: Waveshare 7.3inch ACeP 7-Color E-Paper E-Ink Display Module

mendip-defender, 2023-09-07:
Any chance there is a way to get it to work with this,
7.3inch ACeP 7-Color E-Paper E-Ink Display Module, 800×480 Pixels, SPI Communication
https://www.waveshare.com/product/displays/e-paper/epaper-1/7.3inch-e-paper-hat-f.htm

lmarzen, 2023-09-07:
I dont think it would be too difficult.

This is the epaper driver library used for this project. The 7.3 7-color appears to be supported.
https://github.com/ZinggJM/GxEPD2

I dont have that panel so i can't test it, but if you get it working then feel free to open a pull request and we can merge it into the main project.

lmarzen, 2023-09-07:
If you want I can provide a branch that might work for that display and you can test it and let me know if it works. Let me know if you're interested.

mendip-defender, 2023-09-08:
Yes, please. I will give it a go. I'm a tinker of code, not a coder but I like the project after spotting it on Hackaday.

lmarzen, 2023-09-09:
Try this branch.... 
https://github.com/lmarzen/esp32-weather-epd/tree/43-waveshare-73inch-acep-7-color-e-paper-e-ink-display-module

make sure to set the correct panel in config.h

lmarzen, 2023-09-09:
Let me know how it goes

mendip-defender, 2023-09-10:
[image]

So it works, just only part of the screen shows up.

lmarzen, 2023-09-10:
Okay, that's promising.

I did read more into the epaper driver library source code and for some displays you need to explicitly set it to paged mode.

I pushed some changes try again, and let me know how it goes.
https://github.com/lmarzen/esp32-weather-epd/tree/43-waveshare-73inch-acep-7-color-e-paper-e-ink-display-module

mendip-defender, 2023-09-10:
Bwhahahah I think you have cracked it. When I get a chance I will look to make 7 colour versions of the icons.

[image]

lmarzen, 2023-09-10:
WOOOAAAHH!! success! Ill merge the changes into main.

Multi-colored icons would be a super neat addition as well!

closed 2023-09-10 by lmarzen