lmarzen/esp32-weather-epd #46: Humidity % should be %RH

mendip-defender, 2023-09-10:
This is nit picking but when referring to humidity you should have %RH as its Relative Humidity.

I have seen where to edit it but never done a pull request before so will give that a try after reading the instructions.

lmarzen, 2023-09-10:
Hi @mendip-defender,

While I agree that you are correct that humidity in this case, is actually referring to relative humidity and that %RH is a less ambiguous unit, I think the default % is actually preferable in the weather reporting context.

In the context of public communication and weather reporting, "Humidity" with the "%" symbol is a common and convenient way to express relative humidity. "%RH" is still the technically correct unit to specify relative humidity in scientific and engineering contexts. The choice of notation can vary depending on the context and the level of precision required.

According to wikipedia Relative Humidity (RH) is "usually expressed as a percentage" https://en.wikipedia.org/wiki/Humidity#Relative_humidity
Additionally, every weather reporting website that I checked chose to express Relative Humidity as simply Humidity and use % as the label.
https://forecast.weather.gov/MapClick.php?lat=40.7142&lon=-74.0059
https://openweathermap.org/weather-data
https://www.weatherbug.com/weather-forecast/now/new-york-ny-10001
https://weather.com/weather/today/l/96f2f84af9a5f5d452eb0574d4e4d8a840c71b05e22264ebdc0056433a642c84
https://www.accuweather.com/en/us/new-york/10021/current-weather/14-349727_1_al
https://www.wunderground.com/weather/us/ny/new-york-city
https://www.gismeteo.com/weather-new-york-7190/now/
https://www.yahoo.com/news/weather/united-states/new-york/new-york-2459115
https://www.wetter.com/vereinigte-staaten/new-york-city/US0NY0993.html
https://www.theweathernetwork.com/us/weather/new-york/new-york

Feel free to push back if you still disagree. I am still open to discussing it more, but I need strong reasoning since many people will prefer the % representation.

K20shores, 2023-09-12:
Well, the actual, useful measure of humidity for weather forecasting is specific humidity. Specific humidity is a measure of the water content in the air, typical units might be grams of water vapor for kilogram of air. Percent relative humidity is not a measure of how much water vapor is in the air, it's a measure of how saturated the air is. What the news reports is the percent relative humidity. Pedantically, it would be better to indicate that relative humidity is what's reported, and not specific humidity. 

However, I highly doubt anyone is going to look at the humidity measure of this on your countertop and think "oh, that's clearly specific humidity" because the news only ever talks about relative humidity, even though relative humidity is a slightly useless measure for human comfort. (60% relative humidity at 14 degrees Celsius will feel much better than 30% humidity at 25 degrees Celsius. [numbers from this presentation](http://www.atmo.ttu.edu/schroeder/ATMO_1300/Notes/chapter5.pdf))

If you really wanted to make meteorologists happy, you would read the pressure and convert from relative humidity to specific humidity, and you would display and report the pressure in units of millibars or hectopascals, and temperature would be reported in Celsius or Kelvin.

lmarzen, 2023-09-12:
Thanks for the additional context. Weather forecasting/meteorology is outside of my domain of knowledge. Do you think some people may find it useful to have an option to display specific humidity? There already exist options for unit selection, it should only be minor work to add this option.

K20shores, 2023-09-12:
I mean, if I had the option, I would display that. [Conversion is easy](https://earthscience.stackexchange.com/a/5077/21800). Copied here for brevity:

*What we want:* specific humidity, $q=\frac{w}{w+1}$
*What we have:* percent relative humidity, $\\%RH=100\cdot \frac{e_v}{e_s}$

$e_s$ can be calculated from the [Clausius-Clapeyron equation](https://en.wikipedia.org/wiki/Clausius%E2%80%93Clapeyron_relation). The equation below has the latent heat of vaporization and gas constant of water substituted in (from Atmospheric Thermodynamics: Elementary Physics and Chemistry, North & Erukhimova (2009),  page 110).
```math
\begin{align}
e_s(T)=2.497\times10^9e^{-5417/T} (hPa)
\end{align}
```
So, given $T$ in Kelvin, we can get $e_s$ and then find $e_v=\\%RH\cdot e_s \cdot 10^{-2} (hPa)$. Once we have that, and the current pressure reading $p$, $w$ can be calculated from this equation

```math
w=\frac{e_vR_d}{R_v(p-e_v)}=\frac{287\cdot e_v}{461(p-e_v)}.
```

At this point, we can calculate $q$.

The only potential pitfalls are to ensure that $p$ and $e_v$ have the same units and that we calculate using Kelvin ($K=^\circ C+273.15$). 100 pascals = 1 hectopascal

---
$e_s$=saturation vapor pressure $hPa$
$e_v$=vapor pressure of water $hPa$
$m_v$=mass of water vapor
$m_d$=mass of dry air
$q$=specific humidity
$R_d$=gas constant for dry air = $287.0 J K^{-1} kg^{-1}$
$R_v$=gas constant for water vapor= $461.0J K^{-1} kg^{-1}$
$T$=temperature in $K$
$w$=mixing ratio

To display pressure in millibars, conveniently, 1 millibar = 1 hectopascal.

lmarzen, 2023-09-12:
Thanks for sharing. Feel free to submit a PR if you have time :)

mendip-defender, 2023-09-13:
Sorry, I've been away with work for the first part of the week. I will take a look at the maths and see if I can get it to work as it would be nice to have options based on the data at hand.

My issue is I work in metrology and one of the ISO:17025 documents talks about ambiguity in interpretation when not defining values correctly. The issue is if you had a reading of 88 %RH and you had a measurement uncertainty of 2% you need to be clear is it 2% of 88 %RH or 2 %RH added to the 88 %RH? I know we are not dealing with uncertainties here but as someone who spends far too much time looking at documents I like things to be clear on what they are.

lmarzen, 2023-09-13:
@mendip-defender I don't know the first thing about meteorology, but that's a good point. It seems acceptable to have an option to indicate whether to use the more precise %RH label over the default %.

K20shores, 2023-11-08:
@mendip-defender that may not be incorrect. The units in this are typically g/kg, so it should be small. If you can, open a PR and I can double check everything for you.

mendip-defender, 2023-11-08:
> @mendip-defender that may not be incorrect. The units in this are typically g/kg, so it should be small. If you can, open a PR and I can double check everything for you.

I thought I would just publish the maths on my own github,
https://github.com/mendip-defender/conversion_calculators/blob/main/Specific%20Humidity%20from%20Relative%20Humdity.ipynb

K20shores, 2023-12-30:
@mendip-defender your math seems correct to me. It's a small number because it represents that grams of water vapor per unit mass of air. The unit mass of air includes dry air plus moist air. Naturally, it will have a low value