lmarzen/esp32-weather-epd #47: 5v version or 3.3v for the BME280 sensor?

hatute, 2023-09-12:
In the official wiring diagram, the BME280 is connected to a 3.3v power output, but when I am going to put the order, I found that the model on it need the 5v power input and there is also a BME280 with 3.3v. I just want to make sure which version of the BME280 should we use? Or both of them work?

[image]
[image]
[image]

lmarzen, 2023-09-12:
I've tried it with both the 4pin and 6pin versions and have had no problems.

The BME280 itself needs a voltage supply between 1.71V to 3.6V. https://www.mouser.com/datasheet/2/783/BST-BME280-DS002-1509607.pdf

This 4-pin one exposes the I2C interface only (no SPI), but it includes a voltage regulator and MOSFET logic level shifters and will operate at either 5V or 3.3V.  Really the 4-pin version can operate between 1.71V and 5V.

hatute, 2023-09-12:
> I've tried it with both the 4pin and 6pin versions and have had no problems.
> 
> The BME280 itself needs a voltage supply between 1.71V to 3.6V. https://www.mouser.com/datasheet/2/783/BST-BME280-DS002-1509607.pdf
> 
> The 6-pin one exposes both the I2C and SPI Interfaces. It can operate between 1.71V and 3.6V.
> 
> The 4-pin one exposes the I2C interface only (no SPI), but it includes a voltage regulator and MOSFET logic level shifters and will operate at either 5V or 3.3V. Really the 4-pin version can operate between 1.71V and 5V.
> 
> TLDR: you can buy either and it will work just fine :)

Thank you for the thorough explanation, it totally solved my confusion ;)

closed 2023-09-12 by hatute