lmarzen/esp32-weather-epd #48: Partial refresh support

Velkas, 2023-09-18:
I just got all the components (finally!) and got my own unit working as expected and customized for my location/preferences.

Works great!

I was doing some poking around and am having trouble understanding how to get partial refresh working. I was hoping to further enhance that low power goals of this project by preventing the full refresh which takes multiple seconds to occur as it flashes the full screen multiple times to redraw.

I know my panel supports partial refresh according to the spec, but it seems like nothing I change in the display init flow makes any difference -- I still get the full refresh every time.

I have been searching around and have run into dead ends. Are there any resources available or hints as to where I can investigate further in an attempt to implement this?

lmarzen, 2023-09-18:
I remember investigating the partial refresh back 2 years ago or so. There are some examples in the GxEPD2 epaper driver library repository.

I only experimented with the GxEPD2 library. I was able to get partial refreshes working however when the esp32 goes into deep sleep and wakes up again you must reinitialize the screen with a full refresh before you can draw any partial refreshes.

lmarzen, 2023-09-18:
I think it would be preferable to use partial refreshes as you suggest with the occasional full refresh to clean up ghosting effects, but I never implemented it since I wasn't able to get the partial refreshes to happen without a full refresh after deep sleep. It may be possible to get working correctly (i don't know). Definitely something worth revisiting. Let me know if you find anything interesting.

lmarzen, 2023-09-18:
> I know my panel supports partial refresh according to the spec, but it seems like nothing I change in the display init flow makes any difference -- I still get the full refresh every time.

This is the same issue I ran into and was describing in my earlier replies.

Velkas, 2023-09-18:
Thank you for the info, I'll continue digging!

> ... however when the esp32 goes into deep sleep and wakes up again you must reinitialize the screen with a full refresh before you can draw any partial refreshes.

I do wonder, can you elaborate more on why you think this was? I suspect I was running into the same thing. One of the things I thought about trying was to write more state to non-volatile storage as a means of getting around some of the deep-sleep / wakeup limitations. Space is quite limited but at the very least it could help to reduce the amount of elements that need to be refreshed (assuming partial refresh is working).

lmarzen, 2023-09-18:
Not sure the exact reason.

Check out the init documentation from the driver library
https://github.com/ZinggJM/GxEPD2/blob/master/src/GxEPD2_BW.h

This seems to imply that the display must remain powered when esp32 is in deep sleep to avoid full refresh.

'''
    // init method with additional parameters:
    // initial false for re-init after processor deep sleep wake up, if display power supply was kept
    // this can be used to avoid the repeated initial full refresh on displays with fast partial update
    // NOTE: garbage will result on fast partial update displays, if initial full update is omitted after power loss
    // reset_duration = 10 is default; a value of 2 may help with "clever" reset circuit of newer boards from Waveshare 
    // pulldown_rst_mode true for alternate RST handling to avoid feeding 5V through RST pin
    void init(uint32_t serial_diag_bitrate, bool initial, uint16_t reset_duration = 10, bool pulldown_rst_mode = false)
'''

lmarzen, 2023-09-18:
Maybe the data sheet can help shed some light?
https://www.waveshare.com/w/upload/6/60/7.5inch_e-Paper_V2_Specification.pdf

nglessner, 2023-09-18:
Page 18 matches up with what I remember from when I was knee deep into this
project. I would guess that a partial refresh is possible *without *a full
reset so long as the display is kept in deep sleep versus being turned
off.  I also never got so far as to test this though.
[image: image.png]

lmarzen, 2023-09-18:
@nglessner Good find.

I think the solution may be to use `hibernate()` instead of `poweroff`

```C++
// turns powerOff() and sets controller to deep sleep for minimum power use, ONLY if wakeable by RST (rst >= 0)
    void hibernate()
```

lmarzen, 2023-09-18:
@nglessner Good find.

GxEPD2 provides a `hibernate()` function which may acheive this?
```C++
// turns powerOff() and sets controller to deep sleep for minimum power use, ONLY if wakeable by RST (rst >= 0)
    void hibernate()
```

nglessner, 2023-09-18:
I think the only way to know for sure is to test it. Finding (and trusting)
any documentation beyond the datasheet you linked above has been difficult.

lmarzen, 2023-09-18:
You're right.

I did some more reading though, and found some insight from the author of the GxEPD2 driver library.
https://forum.arduino.cc/t/waveshare-e-paper-displays-with-spi/467865/1244?page=63

> GxEPD2 does support deep sleep for controllers that support deep sleep, through method hibernate().
> 
> The controller IL3820 of the panel GDEH029A1 supports "deep sleep mode" command 0x10, used by the method hibernate() of the library GxEPD2. I measured about 0.8 micro amperes during hibernate, compared to about 22 micro amperes during normal powerOff(). Maximum power measured during refresh was about 3mA, outside refresh 800 micro amps. I measured with a Waveshare 2.9 b/w board.
> 
> It seems ridiculous to ask for deep sleep of the e-paper when looking at your other hardware and its power consumption. I hope you protect the data lines of your e-paper from the 5V of the MEGA IO pins!

lmarzen, 2023-12-17:
Stale. Will reopen if discussion resumes.

closed 2023-12-17 by lmarzen

ltwin8, 2024-08-18:
would it be possible to use the RTCs RAM or some other non volatile ram that remains powered when in sleep to store the current screen (not as pixels but information to build it) then if woken up just check where to does the screen need to change?