lmarzen/esp32-weather-epd #51: Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.

Ianszh, 2023-09-23:
Hello and thanks for the cool project.

After solving a USB Serial problem I managed to compile and upload the code.
But I now have another problem, it get's into a reboot loop because of this error

`Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled`

Here's the complete output

`ELF file SHA256: dee408f393c3a310

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Battery voltage: 4.14
Connecting to '*******'.........................................
IP: 192.168.0.44
Saturday, September 23, 2023 15:13:34
Attempting HTTP Request: api.openweathermap.org/data/2.5/onecall?lat=47.4222&lon=8.3639&lang=en&units=standard&exclude=minutely&appid={API key}
  200 OK
Attempting HTTP Request: api.openweathermap.org/data/2.5/air_pollution/history?lat=47.4222&lon=8.3639&start=1695388418&end=1695474817&appid={API key}
  200 OK
Reading from BME280... Success
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40176c2c: 00004136 00004136 1d004136
Core  1 register dump:
PC      : 0x40176c32  PS      : 0x00060f30  A0      : 0x800db446  A1      : 0x3ffd8a80  
A2      : 0x3ffc4188  A3      : 0x020036d1  A4      : 0x3ffd8acc  A5      : 0x00000000  
A6      : 0x3ffd8a7c  A7      : 0x3ffc5820  A8      : 0x3f44ace8  A9      : 0x3ffd8a40  
A10     : 0x00000003  A11     : 0x401775e8  A12     : 0x00000045  A13     : 0x0000015c  
A14     : 0x3ffd8a78  A15     : 0x3ffd8a7a  SAR     : 0x00000010  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4008bc75  LEND    : 0x4008bc85  LCOUNT  : 0xffffffff  

Backtrace: 0x40176c2f:0x3ffd8a80 0x400db443:0x3ffd8aa0 0x400d99a1:0x3ffd8b40 0x400e72de:0x3ffd8d20`

Did I mess anything up in the config.cpp or other settings?

Thanks

lmarzen, 2023-09-23:
What epaper panel are you using?

Ianszh, 2023-09-23:
Found the problem the "C" was missing in `#define EUROPE_CAQI`

But, my screen is not showing anything, I've gone over the screen connections a few times and I'm sure they're correct?

Ianszh, 2023-09-23:
> What epaper panel are you using?

a Waveshare 7.5-inch HD e-paper (B)

`/ E-PAPER PANEL
// This project supports the following E-Paper panels:
//   DISP_BW_V2 - Waveshare 7.5in e-paper (v2)      800x480px  Black/White
//   DISP_3C_B  - Waveshare 7.5in e-Paper (B)       800x480px  Red/Black/White
//   DISP_7C_F  - Waveshare 7.3in ACeP e-Paper (F)  800x480px  7-Color
//   DISP_BW_V1 - Waveshare 7.5in e-paper (v1)      640x384px  Black/White
// Uncomment the macro that identifies your physical panel.
// #define DISP_BW_V2
#define DISP_3C_B
// #define DISP_7C_F
// #define DISP_BW_V1

// 3 COLOR E-INK ACCENT COLOR
// Defines the 3rd color to be used when a 3+ color display is selected.
#if defined(DISP_3C_B) || defined(DISP_7C_F)
  // #define ACCENT_COLOR GxEPD_BLACK
  #define ACCENT_COLOR GxEPD_RED
  // #define ACCENT_COLOR GxEPD_GREEN
  // #define ACCENT_COLOR GxEPD_BLUE
  // #define ACCENT_COLOR GxEPD_YELLOW
  // #define ACCENT_COLOR GxEPD_ORANGE
#endif
`

lmarzen, 2023-09-23:
Hmmm, the output looks promising. Are you using the default pinout from config.cpp?

Ianszh, 2023-09-23:
> Hmmm, the output looks promising. Are you using the default pinout from config.cpp?

I did pinouts per the wiring_diagram.png which looks the same

lmarzen, 2023-09-23:
Maybe a recent change is causing issues for that panel.

Try the following.
Go to renderer.cpp and find the function initDisplay()

You will see the following lines...
```C++
  // display.fillScreen(GxEPD_WHITE);
  display.setFullWindow();
  display.firstPage();
```

Change this to 
```C++
  display.fillScreen(GxEPD_WHITE);
  display.setFullWindow();
  // display.firstPage();
```

Ianszh, 2023-09-23:
Still the same I'm afraid.

Does GxEPD2 support this panel?
Should I change the height in renderer.h to 528?

`#ifdef DISP_3C_B
  #define DISP_WIDTH  800
  #define DISP_HEIGHT 480
  #include <GxEPD2_3C.h>
  extern GxEPD2_3C<GxEPD2_750c_Z08,
                   GxEPD2_750c_Z08::HEIGHT / 2> display;`

lmarzen, 2023-09-23:
Do you have the 800×480 or 880×528 version. this project only has support for the 800x480 version

Ianszh, 2023-09-23:
Unfortunately the 880x528 Discontinued version 😢

https://www.waveshare.com/7.5inch-hd-e-paper-b.htm

lmarzen, 2023-09-23:
I think GxEPD2 supports that panel so it's possible to add support for this project however there is no scaling support for the display so only part of the display would be occupied.

Ianszh, 2023-09-23:
Ok thanks, looks like I'll have to get a 800x480 version and do something else with this one.

One other question or I could start another issue.
How about an option for a 2nd BME280, local outside temp etc. data sent via a MQTT broker?

lmarzen, 2023-09-23:
I am no opposed to that feature, however I do not have the time to develop that right now.

Ianszh, 2023-09-23:
Ok thanks, if you ever do have time, would be interesting

closed 2023-09-23 by lmarzen