lmarzen/esp32-weather-epd #52: The Red/Black/White e-paper is only visible while refreshing

K20shores, 2023-09-25:
I have everything connected. When the controller refreshes the screen, I see a white background with the black text briefly, and then the display goes dark. The information is written on the screen, but it's like there is a nearly opaque darkish-red color on the screen. I've tried to include some pictures.

Any ideas on what might be going on?

White refresh:
[image]

Red-ish-dark-ish opaqueness:
[image]

K20shores, 2023-09-25:
Never mind. You already covered this. I had to uncomment `DISP_3C_B` in config.h, of course.

However, it does seem to flash quite a bit, as if it's refreshing several times per second before it stabilizes. Not sure if that's expected or not

lmarzen, 2023-09-25:
Glad you were able to get it figured out. And yes that is the expected behavior. The waveshare color papers require quite a bit of 'flashing' to refresh.

closed 2023-09-25 by K20shores