lmarzen/esp32-weather-epd #54: 75b_V2 display is not working

bob-u, 2023-10-06:
I have a no name tricolor display that works with waveshare arduino examples, when 75b_v2 is selected.  It also works with ESPHome.  However esp32-weather-epd does nothing - screen does not flicker, and screen init / render portion happens very quickly which leads me to believe that this portion fails quietly (no errors in console) 

I added print to console when initDisplay() and drawString() function is called, output below.  Any suggestions on  how to troubleshoot this issue?  I'm pretty sure that pin assignment is correct and as I mentioned earlier my setup works with other apps.

```
Attempting HTTP Request: api.openweathermap.org/data/2.5/air_pollution/history?lat=42.3318&lon=-71.1212&start=1696514055&end=1696600454&appid={API key}
  200 OK
Reading from BME280... BME not found
initDisplay
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
_PowerOn : 26
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
drawString
_Update_Full : 10
_PowerOff : 2
_PowerOff : 2
Awake for 12.064s
Deep-sleep for 352s
```

bob-u, 2023-10-07:
Had to swap SCK with MOSI and use hspi.  Great looking weather dashboard, thank you for making it.  What would make it even better would be a way to pull sensor data from HA.

closed 2023-10-07 by bob-u

torytyler, 2023-12-31:
I'm having the same issue with the E-Paper Driver ESP32 Board, the pinout is using hspi and not posting data to the e-paper display.

Is there any way you can provide the changes you made? This display is working with the waveshare arduino examples. So I know it's gotta be something simple, not a complete rewrite