lmarzen/esp32-weather-epd #55: Failed to connect to ESP32

DaveTakesPictures, 2023-10-07:
I'm having trouble uploading the project. There is one warning early on, and though I just updated Python, it persists and I'm not sure if that's the cause anyway.

Second error, rather generic, I looked at the link, tried suggestions such as disconnecting everything and still can't upload. It's my first time using PIO and just guessing... shouldn't I have a USB connection? I see /dev/cu.Bluetooth-Incoming-Port but I'm not connected by bluetooth?? I'm on a mac running OS12.7 and VS 1.83.0

A fatal error occurred: Failed to connect to ESP32: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
*** [upload] Error 2

Full terminal ouput below:

> ================================================== [FAILED] Took 92.01 seconds ==================================================

Environment                 Status    Duration
--------------------------  --------  ------------
dfrobot_firebeetle2_esp32e  FAILED    00:01:32.008
============================================= 1 failed, 0 succeeded in 00:01:32.008 =============================================

 *  The terminal process "platformio 'run', '--target', 'upload'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

 *  Executing task in folder platformio: platformio run --target upload 

/Users/davepixmb12/.platformio/penv/lib/python3.9/site-packages/urllib3/__init__.py:34: NotOpenSSLWarning: urllib3 v2.0 only supports OpenSSL 1.1.1+, currently the 'ssl' module is compiled with 'LibreSSL 2.8.3'. See: https://github.com/urllib3/urllib3/issues/3020
  warnings.warn(
Processing dfrobot_firebeetle2_esp32e (board: dfrobot_firebeetle2_esp32e; platform: espressif32 @ 6.4.0; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------
/Users/davepixmb12/.platformio/penv/lib/python3.9/site-packages/urllib3/__init__.py:34: NotOpenSSLWarning: urllib3 v2.0 only supports OpenSSL 1.1.1+, currently the 'ssl' module is compiled with 'LibreSSL 2.8.3'. See: https://github.com/urllib3/urllib3/issues/3020
  warnings.warn(
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/dfrobot_firebeetle2_esp32e.html
PLATFORM: Espressif 32 (6.4.0) > DFRobot Firebeetle 2 ESP32-E
HARDWARE: ESP32 80MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 40 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Adafruit BME280 Library @ 2.2.2
|-- Adafruit BusIO @ 1.14.4
|-- Adafruit Unified Sensor @ 1.1.13
|-- ArduinoJson @ 6.21.3
|-- GxEPD2 @ 1.5.2
|-- pollutant-concentration-to-aqi
|-- HTTPClient @ 2.0.0
|-- SPI @ 2.0.0
|-- WiFi @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
|-- Preferences @ 2.0.0
|-- Wire @ 2.0.0
Building in release mode
Retrieving maximum program size .pio/build/dfrobot_firebeetle2_esp32e/firmware.elf
Checking size .pio/build/dfrobot_firebeetle2_esp32e/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  31.0% (used 101744 bytes from 327680 bytes)
Flash: [====      ]  44.5% (used 1401097 bytes from 3145728 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: /dev/cu.Bluetooth-Incoming-Port
Uploading .pio/build/dfrobot_firebeetle2_esp32e/firmware.bin
esptool.py v4.5.1
Serial port /dev/cu.Bluetooth-Incoming-Port
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
*** [upload] Error 2
================================================== [FAILED] Took 95.12 seconds ==================================================

Environment                 Status    Duration
--------------------------  --------  ------------
dfrobot_firebeetle2_esp32e  FAILED    00:01:35.118
============================================= 1 failed, 0 succeeded in 00:01:35.118 =============================================

 *  The terminal process "platformio 'run', '--target', 'upload'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

lmarzen, 2023-10-08:
Try running  `ioreg -p IOUSB` which will list USB devices. This way you can see if it is being detected by the OS.

DaveTakesPictures, 2023-10-09:
Thanks Luke, I'll try and work on the Mac connection later, I wrangled a PC and was able to connect and upload successfully! Works like a charm! Thanks for the great project!

closed 2023-10-20 by lmarzen