lmarzen/esp32-weather-epd #61: One Call API 3.0

diettelo, 2023-10-23:
Hello,

i need some help with your Weather Station Project.

This is my Api Key and i also signed to 3.0, everything working:
https://api.openweathermap.org/data/3.0/onecall?lat=47.28&lon=12.83&units=metric&lang=de&appid=e186e663cc7f8efc0006a3129edabc25

but when i compile everything and upload my Firmware, Allway One Call 3.0 API -1: connection refused.

Here comes the funny part:

In my router it i can see an Ip Adress also give it a free Static IP via Mac Adress.
I can Ping that Adress but Target host not avaible.

I can send you my Code if you whant to take a look.
I have no clue what is wrong.

I also tried to change to 2.5 API Link and Version, nothing.

best regards,

diettelo
[image]
[image]
[image]
[image]
[image]
[image]

lmarzen, 2023-10-23:
Use the default endpoint.

diettelo, 2023-10-23:
Sorry, can you show me a little bit more.
What Endpoind do you mean?

Have you an example?

lmarzen, 2023-10-23:
Revert your OWM_ENDPOINT setting back to "api.openweathermap.org". This setting is here because some paid tiers have special endpoints, but usually the endpoint is just "api.openweathermap.org".

diettelo, 2023-10-23:
Sorry, can you show me a little bit more.
What Endpoind do you mean?

Have you an example?
    

edit:

i try it, thx.

lmarzen, 2023-10-23:
The code will construct the entire request based on your settings, so no need to specify https://api.openweathermap.org/data/3.0/onecall?lat=47.28&lon=12.83&units=metric&lang=de&....

Also if "e186e663cc7f8efc0006a3129edabc25" is your real api key you just leaked it. If this is the case, go to your openweathermap account to generate a new one and delete the old one.

diettelo, 2023-10-23:
Thx, i will change that.

closed 2023-10-23 by lmarzen

diettelo, 2023-10-24:
Thanks for your Support.
It is running very fine.

The only Thing i wanted to know is:
Can the Small Clock larger be displayed and Postion Upper Left, where the Rain Symbol is?

lmarzen, 2023-10-24:
There isn't a setting for that, but you could do that, by changing the position of where elements are drawn in renderer.cpp.

diettelo, 2023-10-24:
Ok Thanks, I´ll check it out.

Is there a way to Support your work?

lmarzen, 2023-10-25:
Are you asking to support the project with contributions, donating hardware, financially?
All are welcome and would be greatly appreciated. If you wanted to support the project with a donation I could setup the github sponsorships for this repo. I started this project for fun and I am glad that people are taking interest like yourself and want to support the project.