lmarzen/esp32-weather-epd #63: What do I need to change to debug and upload with a different version of esp32?

PaekHyun, 2023-11-04:
I have not dfrobot_firebeetle2_esp32e.
but this one .
https://www.waveshare.com/wiki/E-Paper_ESP32_Driver_Board

What do I need to change to debug and upload with a different version of esp32?

please help me.
thank you.

lmarzen, 2023-11-04:
You will need to change the board in the platform.ini file and change your pinnouts if necessary in config.cpp

PaekHyun, 2023-11-04:
do you know where is T2 ADC pin with this esp32 board?
https://www.waveshare.com/wiki/E-Paper_ESP32_Driver_Board

maybe it doesnt connect battery adc pin

[image]

PaekHyun, 2023-11-04:
or how to disabled low batter mode?

lmarzen, 2023-11-04:
It might be IO2

lmarzen, 2023-11-04:
I'm adding an option to disable battery monitoring now. Give me 10 minutes. I've been meaning to add an option for this for a long while.

lmarzen, 2023-11-04:
I have added an option for disabling battery monitoring in config.h. 675d5f1

PaekHyun, 2023-11-05:
> I have added an option for disabling battery monitoring in config.h. [675d5f1](https://github.com/lmarzen/esp32-weather-epd/commit/675d5f17e033cc8b81fe80d16e1500035324f883)

thank you very much. i'll try it again..

PaekHyun, 2023-11-05:
how do you think, why is there no response on e paper?

serial monitor showed below picture
[image]

what mean of serial monitor message with power on /off, update_full and number?

lmarzen, 2023-11-05:
Those messages are from the epaper driver library and indicate the time each operation took. The times shown are way too short. I eould double check the following...

Display Config: Set switch to position B.

Interface Config: Set switch to position 0.

You epaper pins in config.cpp.

PaekHyun, 2023-11-06:
ohh.......at last i found.
renderer.cpp file

SPI.begin code must be at the top of init display function.
[image]

it works.!!

but i dont know why, it showed weather is so different with here..

PaekHyun, 2023-11-06:
ah my mistake., 
i input wrong latitude.
thank you very much.

closed 2023-11-19 by lmarzen

YuriStruszczynski, 2023-12-17:
> ohh.......at last i found. renderer.cpp file
> 
> SPI.begin code must be at the top of init display function. ![image](https://private-user-images.githubusercontent.com/38004738/280729674-c70c6901-4287-4efe-adcc-4210e611fbea.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI4NDg5ODUsIm5iZiI6MTcwMjg0ODY4NSwicGF0aCI6Ii8zODAwNDczOC8yODA3Mjk2NzQtYzcwYzY5MDEtNDI4Ny00ZWZlLWFkY2MtNDIxMGU2MTFmYmVhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjE3VDIxMzEyNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI4NTQyYjkzMjBlNjJkMjdmNmRhNzRiZjE4ZTRjNmU5NTQ1MTlkOTQ3YTQxODc5OTY1MDA1NDM3ZDYxYTc1ZGMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.uKJ_zZ1CzNMI_9ONdZwIvz8nufLGPrw-w7s0Kqbun5s)
> 
> it works.!!
> 
> thank you!!

Nice! May you also post your PIN config please?

PaekHyun, 2023-12-28:
// Pins for Waveshare e-paper Driver Board

const uint8_t PIN_EPD_BUSY = 25;
const uint8_t PIN_EPD_CS   = 15;
const uint8_t PIN_EPD_RST  = 26;
const uint8_t PIN_EPD_DC   = 27;
const uint8_t PIN_EPD_SCK  = 13;
const uint8_t PIN_EPD_MISO = 12; // 19 Master-In Slave-Out not used, as no data from display
const uint8_t PIN_EPD_MOSI = 14;

pin numbers are wavehare e epaper driver board default setting.