lmarzen/esp32-weather-epd #66: Add Sensor Module

bqbowden, 2023-11-22:
Would it be possible with this configuration to add a sensor module gather data from a location other than the main station?

lmarzen, 2023-11-22:
That's not a currently supported feature.

lmarzen, 2023-12-17:
@cwitting,

I am not that familiar with MQTT. It seems like there might be interest in this feature and I would consider merging this into the project on the condition that documentation is also updated to explain how to enable and use this feature.

lmarzen, 2024-02-16:
Closed since this issue is inactive. If community interest returns I am happy to discuss this feature again in the future.

PatrickGlatz, 2024-03-26:
Hi @lmarzen, Hi @cwitting,

i really would love to see measurement data to be pulled from MQTT. MQTT as one of the IoT Standards out there is perfect for this, as it offers to keep the last pushed value as long as a new value is published. That means, it doesn't matter when the weather-epd is waking up and fetches the data, it is there! One would have to configure the MQTT topics with the "Retained"-Flag, simple as that. 

That would also give hundreds of additional options. Showing measurement data from a different location, ... 

Here are some informational links:
https://en.wikipedia.org/wiki/MQTT
https://www.hivemq.com/mqtt/

My usecase would be: I do have a more advanced weather station at home, indoor + outdoor 
https://www.ecowitt.com

There are some semi professionals who are sharing there knowledege here (sorry, most german):
https://wetterstationsforum.info/

Would be happy if we could share our knowledge here.

Thanks!

lmarzen, 2024-03-26:
Hi @PatrickGlatz,

Thanks for sharing some resources on MQTT. I am in agreeance that MQTT would be a great addition to the project.

Unfortunately, I currently do not have much time to commit to implementing new features as I am busy with school and research. With that said, if you or anyone else are willing to implement the feature yourself, I would be more than happy to merge it into the project. 

Thank you,

Luke

PatrickGlatz, 2024-03-27:
Hi @lmarzen,

i will try to get the necessary parts. Though, i would like to display even more information, especially local rain volumes, which is super easy and always correct with an own weather station. As soon as i have the parts, i will try to add the mqtt section.

Thanks
Patrick

cwitting, 2024-04-01:
Hi @PatrickGlatz and @lmarzen 

I did some work to get MQTT to work for the humidity, you can see how I did it here: https://github.com/cwitting/esp32-weather-epd/commit/06c64453a2b24892934bf875cd09d431a70e3240

However it is quite hacky right now, and I do not have the time to clean it up. But I think it should be easy to generalize it for other fields.

closed 2024-08-25 by lmarzen