lmarzen/esp32-weather-epd #71: Waveshare e-paper driver hat v.2.2 - different wiring

martinriegel, 2023-12-26:
Hi. 
I got a Waveshare e-paper driver hat v.2.2 with my 7.5 e-paper hat (B) display.

I had to wire the VCC to the 3V3 pin on the ESP to get the display up and running. So maybe add that as an example to the wiring guide. 
[image]

PaekHyun, 2023-12-28:
i think you can change display config.
how about display config change to "A"

lmarzen, 2023-12-28:
@martinriegel
Thanks for sharing I may add this.

@PaekHyun 
You can, and it may work however, you may lose some contrast or notice some graininess/noise, especially around the border.

Additionally, waveshare ([waveshare-driver-hat](https://www.waveshare.com/wiki/E-Paper_Driver_HAT)) and good-display ([despi-c02](https://v4.cecdn.yun300.cn/100001_1909185148/DESPI-C02%20Connector%20Board%20for%20E-paper%20Display%20V1.1.pdf)) indicate that the displays used by this project should be set to .47R (B). I saw some artifacting myself when setting it to 3R (A) so I wouldn't recomment this.

closed 2023-12-28 by lmarzen