lmarzen/esp32-weather-epd #73: Display won't turn on

xJBx1, 2024-01-14:
Hi, 
I have a 7.3in F e paper display and I have modified the code so that I get to see other weather options in place of the two indoor ones since I don't have the sensor, compiled the code and it builds wells and uploads to the esp32 (I have a regular dev board not the firebeetle), but when I run the sketch on the esp I get time, wifi, battery voltage, api response OK, power, reset and deepsleep seconds. It completely skips over the updating the display part and I'm confused because I have done the modifications for the display in the config files.

Can anyone help me on this?

lmarzen, 2024-01-15:
Can you please share your serial monitor output?

Also, did you update the board configuration in platform.ini?

xJBx1, 2024-01-15:
[image]

that is the output I'm getting 

[platformio]
default_envs = esp32dev
; default_envs = firebeetle32

[env:esp32dev]
board = esp32dev
monitor_speed = 115200
; override default partition table
; https://github.com/espressif/arduino-esp32/tree/master/tools/partitions
board_build.partitions = huge_app.csv
; change MCU frequency, 240MHz -> 80MHz (for better power efficiency)
board_build.f_cpu = 80000000L
 
the above code is from my platform.ini file.....

Since the display is turning on, would it be an issue with the draw functions??
on the waveshare examples to draw they have codes like SendCommand(0x77) etc. which is not present here, would that make a difference?

lmarzen, 2024-01-15:
What driver board are you using? And can you confirm the switches are correctly set?

xJBx1, 2024-01-15:
I am using a XCLUMA branded esp32 cp2102, it is a normal 30 pin esp32 board, I have 8 connectors for the e-paper, 3v3, gnd, bsy, rst, cs, dc, din, sck. So I had connected 3v3 and gnd to their respective pins, I assume DIN is MOSI and linked it respectively on vs, and i have left PWR and MISO as it is since I am not using them.

const uint8_t PIN_EPD_BUSY =  4; // 5 for micro-usb firebeetle
const uint8_t PIN_EPD_CS   =  5; 
const uint8_t PIN_EPD_RST  = 21;
const uint8_t PIN_EPD_DC   = 22;
const uint8_t PIN_EPD_SCK  = 18;
const uint8_t PIN_EPD_MISO = 19; // 19 Master-In Slave-Out not used, as no data from display
const uint8_t PIN_EPD_MOSI = 23;
const uint8_t PIN_EPD_PWR  = 14; // Irrelevant if directly connected to 3.3V

Those are my connections, these connections worked when I want to display just a few photos on the e-paper as a test from the arduino ide. 

Please suggest if these are correct. 

[image]

That is my board above.

lmarzen, 2024-01-15:
I meant are you using the waveshare hat or the despi?

xJBx1, 2024-01-15:
I am using the hat and I've selected that option in the config as well

xJBx1, 2024-01-15:
Yea, I tried debugging the program and it seems to run through everything as it's meant to, just drawing it on the actual display is the problem, are the EPD functions universal?

lmarzen, 2024-01-15:
You could try some of the things suggested here.
https://github.com/lmarzen/esp32-weather-epd/issues/63

I don't have the board you're using or the panel so I can not easily debug this issue. If any of the things in the before mentioned link helps let me know and we can add the fix to the project.

xJBx1, 2024-01-15:
Alright I'll have a crack at it and let you know

xJBx1, 2024-01-15:
void initDisplay()
{
  SPI.begin(PIN_EPD_SCK,
            PIN_EPD_MISO,
            PIN_EPD_MOSI,
            PIN_EPD_CS);
  
  display.init(115200, true, 2, false);
  display.setRotation(0);
  display.setTextSize(1);
  display.setTextColor(GxEPD_BLACK);
  display.setTextWrap(false);
  display.fillScreen(GxEPD_WHITE);
  display.setFullWindow();
  // display.firstPage(); // use paged drawing mode, sets fillScreen(GxEPD_WHITE)
  return;
} // end initDisplay

made that the init function but still no luck :(((

are there any other fixes for this?

_InitDisplay reset : 21
_PowerOn : 2
_refresh : 21
_PowerOff : 2
Awake for 16.700s
Deep-sleep for 2725s
that's what the serial monitor looks like

xJBx1, 2024-01-16:
Hi,
I am adding a link to the modified code I have, would you be able to review that please and let me know what I have done wrong?
https://github.com/xJBx1/weather_module

xJBx1, 2024-01-17:
Hey Luke,
Had a quick one, would moving the platformio.ini file outside the platformio folder help since it will include all the files in the repository?

lmarzen, 2024-01-17:
No moving the platform.ini file will not help. Platformio depends on a certain directory structure such as the src, include, libraries directories on the same level as the .ini file.

lmarzen, 2024-01-17:
Your CI is failing because you renamed `platformio` directory to `Platformio`

xJBx1, 2024-01-17:
I changed the file name to platformio but that didn't work either.....is it somewhere in the code I need to change it?

xJBx1, 2024-01-17:
Did some debugging today, and the part which isn't working is the communication part, I used the display.drawTriangle function which is mutually exclusive to the whole program, and in theory should work however it didn't print anything, so I am suspecting it is the GxEPD which has some errors on it's part since the rest of the code seems to work fine.......What would you recommend I do?

closed 2024-01-29 by xJBx1