lmarzen/esp32-weather-epd #77: Change PIN_EPD_CS number in config.cpp

octopuss, 2024-01-26:
Please could you change default config.cpp https://github.com/lmarzen/esp32-weather-epd/blob/ebfe399e4b00c0c02df7ed245b46f267b9326e57/platformio/src/config.cpp#L26 so it is not set to LED_BUILTIN on FireBeetle 2 ESP32-E (or write some comment there). It is also in the wiring diagrams. It took me too much time to figure out why my display shows random noise. 

Thank you!

lmarzen, 2024-01-26:
Wow I don't know how that hasn't been an issue until now. Great find! I will update this once I get a chance.

octopuss, 2024-01-26:
My guess is that standard 7.3 inch waveshare displays are somehow ok with
that( i have one project done almost a year ago) Now i am creating 2 inch
(400x168 waveshare) reduced version for my desktop. Anyway cool roject.
I've also connected it to Home assistant so it displays my outdoor
temperature sensor values