lmarzen/esp32-weather-epd #79: Battery voltage read is wrong

FrAllard, 2024-02-05:
I recently enabled the option to display the battery voltage on the screen for the Firebeetle ESP32-E v1.0. My concern arose when the voltage reading was unexpectedly high, showing 4.40V, which surpasses the typical maximum charge level for a LiPo battery (4.2V). However, upon verification with a multimeter, the actual battery voltage was confirmed to be safely below 4.2V.

This led me to question the accuracy of the firmware's battery voltage reading. Incorrect voltage readings could potentially compromise the effectiveness of the battery's monitoring and protection systems, possibly failing to trigger the necessary protections due to an overestimated voltage reading.

To address this issue, I adjusted the conversion factor back to (3.3 / 2.0) in the setup function of the main.cpp file, contrary to the comments suggesting a different factor for the ESP32-E version. This adjustment, which leans towards underestimation, is a precautionary measure I prefer; it theoretically triggers protection mechanisms earlier, which is a safer approach for battery management.

It's worth noting that the Firebeetle seems to charge the battery correctly, so my concern about overcharging is minimal. However, this experience has highlighted that there may be variations in the Firebeetle ESP32-E boards concerning how battery voltage is read. This discrepancy is something users should be aware of, especially since it can affect the accuracy of displayed battery voltages and potentially impact battery protection logic.

lmarzen, 2024-02-05:
Thanks, for sharing your concern. I feared this may be the case. Maybe I should make this value adjustable via the config? I'll do some research first and see if there is a better way to go about this. Feel free to share any other suggestions you may have.

-Luke

lmarzen, 2024-02-05:
I'll have to experiment with this, but it seems like the esp32 may be precalibrated by the manufacturer,
https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32/api-reference/peripherals/adc.html#_CPPv426esp_adc_cal_raw_to_voltage8uint32_tPK29esp_adc_cal_characteristics_t

lmarzen, 2024-02-17:
Should be fixed now. Reopen this issue if you are still having issues.

closed 2024-02-17 by lmarzen