lmarzen/esp32-weather-epd #81: Display somehow not turning on

Smutssch, 2024-02-11:
So i recreated the project exactly the same way built here, but after i upload the code the display doesnt show anything, and when I unplug the USB cable the red light goes out (ig thats normal) but there is no green light I expected there to be, because the picture of the running project showed a green light. I´m really new to hardware stuff like this and It could just be something really dumb, but still I would appreciate it a lot if you helped me out here. I´m doing this as a project for school and don´t really want it to fail.

lmarzen, 2024-02-11:
Please share the terminal output.

Smutssch, 2024-02-11:
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/dfrobot_firebeetle2_esp32e.html
PLATFORM: Espressif 32 (6.4.0) > DFRobot Firebeetle 2 ESP32-E
HARDWARE: ESP32 80MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 41 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Adafruit BME280 Library @ 2.2.4
|-- Adafruit BusIO @ 1.15.0
|-- Adafruit Unified Sensor @ 1.1.14
|-- ArduinoJson @ 6.21.5
|-- GxEPD2 @ 1.5.5
|-- pollutant-concentration-to-aqi
|-- HTTPClient @ 2.0.0
|-- SPI @ 2.0.0
|-- WiFi @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
|-- esp32-weather-epd-assets
|-- Preferences @ 2.0.0
|-- Wire @ 2.0.0
Building in release mode
Compiling .pio\build\dfrobot_firebeetle2_esp32e\src\_strftime.cpp.o
Compiling .pio\build\dfrobot_firebeetle2_esp32e\src\api_response.cpp.o
Compiling .pio\build\dfrobot_firebeetle2_esp32e\src\client_utils.cpp.o
Compiling .pio\build\dfrobot_firebeetle2_esp32e\src\config.cpp.o
Compiling .pio\build\dfrobot_firebeetle2_esp32e\src\conversions.cpp.o
Compiling .pio\build\dfrobot_firebeetle2_esp32e\src\display_utils.cpp.o
Compiling .pio\build\dfrobot_firebeetle2_esp32e\src\locales\locale.cpp.o
Compiling .pio\build\dfrobot_firebeetle2_esp32e\src\main.cpp.o
Compiling .pio\build\dfrobot_firebeetle2_esp32e\src\renderer.cpp.o
Compiling .pio\build\dfrobot_firebeetle2_esp32e\libce9\Wire\Wire.cpp.o
Compiling .pio\build\dfrobot_firebeetle2_esp32e\lib2ef\SPI\SPI.cpp.o
Compiling .pio\build\dfrobot_firebeetle2_esp32e\lib5d5\Adafruit BusIO\Adafruit_BusIO_Register.cpp.o
Compiling .pio\build\dfrobot_firebeetle2_esp32e\lib5d5\Adafruit BusIO\Adafruit_I2CDevice.cpp.o
Compiling .pio\build\dfrobot_firebeetle2_esp32e\lib5d5\Adafruit BusIO\Adafruit_SPIDevice.cpp.o
Compiling .pio\build\dfrobot_firebeetle2_esp32e\lib93b\Adafruit Unified Sensor\Adafruit_Sensor.cpp.o
Compiling .pio\build\dfrobot_firebeetle2_esp32e\lib529\Adafruit BME280 Library\Adafruit_BME280.cpp.o
Compiling .pio\build\dfrobot_firebeetle2_esp32e\lib935\Adafruit GFX Library\Adafruit_GFX.cpp.o
Compiling .pio\build\dfrobot_firebeetle2_esp32e\lib935\Adafruit GFX Library\Adafruit_GrayOLED.cpp.o
Archiving .pio\build\dfrobot_firebeetle2_esp32e\libce9\libWire.a
Compiling .pio\build\dfrobot_firebeetle2_esp32e\lib935\Adafruit GFX Library\Adafruit_SPITFT.cpp.o
Archiving .pio\build\dfrobot_firebeetle2_esp32e\lib2ef\libSPI.a
Compiling .pio\build\dfrobot_firebeetle2_esp32e\lib935\Adafruit GFX Library\glcdfont.c.o
Compiling .pio\build\dfrobot_firebeetle2_esp32e\lib9d1\GxEPD2\GxEPD2_EPD.cpp.o
Archiving .pio\build\dfrobot_firebeetle2_esp32e\lib93b\libAdafruit Unified Sensor.a
Archiving .pio\build\dfrobot_firebeetle2_esp32e\lib5d5\libAdafruit BusIO.a
Compiling .pio\build\dfrobot_firebeetle2_esp32e\lib9d1\GxEPD2\epd3c\GxEPD2_1248c.cpp.o
Compiling .pio\build\dfrobot_firebeetle2_esp32e\lib9d1\GxEPD2\epd3c\GxEPD2_154_Z90c.cpp.o
Compiling .pio\build\dfrobot_firebeetle2_esp32e\lib9d1\GxEPD2\epd3c\GxEPD2_154c.cpp.o
Compiling .pio\build\dfrobot_firebeetle2_esp32e\lib9d1\GxEPD2\epd3c\GxEPD2_213_Z19c.cpp.o
Archiving .pio\build\dfrobot_firebeetle2_esp32e\lib529\libAdafruit BME280 Library.a
Compiling .pio\build\dfrobot_firebeetle2_esp32e\lib9d1\GxEPD2\epd3c\GxEPD2_213_Z98c.cpp.o
Compiling .pio\build\dfrobot_firebeetle2_esp32e\lib9d1\GxEPD2\epd3c\GxEPD2_213c.cpp.o
Compiling .pio\build\dfrobot_firebeetle2_esp32e\lib9d1\GxEPD2\epd3c\GxEPD2_266c.cpp.o
Archiving .pio\build\dfrobot_firebeetle2_esp32e\lib935\libAdafruit GFX Library.a
Compiling .pio\build\dfrobot_firebeetle2_esp32e\lib9d1\GxEPD2\epd3c\GxEPD2_270c.cpp.o
Compiling .pio\build\dfrobot_firebeetle2_esp32e\lib9d1\GxEPD2\epd3c\GxEPD2_290_C90c.cpp.o
Compiling .pio\build\dfrobot_firebeetle2_esp32e\lib9d1\GxEPD2\epd3c\GxEPD2_290_Z13c.cpp.o
Compiling .pio\build\dfrobot_firebeetle2_esp32e\lib9d1\GxEPD2\epd3c\GxEPD2_290c.cpp.o
Compiling .pio\build\dfrobot_firebeetle2_esp32e\lib9d1\GxEPD2\epd3c\GxEPD2_420c.cpp.o
[... 294 lines cut ...]
Writing at 0x00113da6... (70 %)
Writing at 0x0011908a... (72 %)
Writing at 0x0011e768... (75 %)
Writing at 0x00123d89... (77 %)
Writing at 0x00129616... (79 %)
Writing at 0x0012f39b... (81 %)
Writing at 0x00134cea... (84 %)
Writing at 0x0013bbe3... (86 %)
Writing at 0x00143dd8... (88 %)
Writing at 0x0014b68f... (90 %)
Writing at 0x001507b1... (93 %)
Writing at 0x00156430... (95 %)
Writing at 0x0015b904... (97 %)
Writing at 0x00160e6d... (100 %)
Wrote 1400560 bytes (718661 compressed) at 0x00010000 in 18.8 seconds (effective 595.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
========================================================================================= [SUCCESS] Took 44.23 seconds =========================================================================================
Environment                 Status    Duration
--------------------------  --------  ------------
dfrobot_firebeetle2_esp32e  SUCCESS   00:00:44.233
========================================================================================= 1 succeeded in 00:00:44.233 ========================================================================================= 
 *  Terminal will be reused by tasks, press any key to close it.

lmarzen, 2024-02-11:
I should have been more specific. Share the terminal monitor output. VSCode has an option on the bottom to view the usb terminal monitor. Click that, then press the reset button on the esp32. This output will help us debug your problem.

Smutssch, 2024-02-11:
Now that I see this, it´s probably because of the API

 *  Executing task: C:\Users\samue\.platformio\penv\Scripts\platformio.exe device monitor 

--- Terminal on COM3 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Battery voltage: 4.03
Connecting to 'Koti'...............................................................................
IP: 192.168.2.175
Waiting for SNTP synchronization..................
Sunday, February 11, 2024 09:34:26
Attempting HTTP Request: api.openweathermap.org/data/3.0/onecall?lat=51.227741&lon=6.773456&lang=de&units=standard&exclude=minutely&appid={API key}
  401 Unauthorized
  401 Unauthorized
  401 Unauthorized

lmarzen, 2024-02-11:
So you have 2 issues. First, your API key is invalid "401" (this won't stop the display from working, but should result in the error message displaying on the screen).  The second issue (which is why your display isn't displaying anything) is that your display may not be wired correctly. "Busy Timeout! _PowerOn : 10000227" Here is another person who had the same problem and suggested that the PWR pin may have been wired incorrectly, https://github.com/lmarzen/esp32-weather-epd/issues/80.

lmarzen, 2024-02-11:
Also double check you have selected the correct driver board in config.h. I'm not sure which one you have, the despi or the waveshare hat.

> // E-PAPER DRIVER BOARD
// Uncomment the macro that identifies your driver board hardware.
#define DRIVER_DESPI_C02
// #define DRIVER_WAVESHARE

lmarzen, 2024-02-11:
Another thing to check is that your driver board has the switch(s) set correctly.

From the readme...

> IMPORTANT: The Waveshare E-Paper Driver HAT has two physical switches that MUST be set correctly for the display to work.
> - Display Config: Set switch to position B.
> - Interface Config: Set switch to position 0.
> 
> IMPORTANT: The DESPI-C02 adapter has one physical switch that MUST be set correctly for the display to work.
> - RESE: Set switch to position 0.47.

Smutssch, 2024-02-11:
Didn´t have the correct driver board selected, but now it just fails to connect to my wifi, although I did change the SSID and the password to my wifi. Also now it shows the battery voltage so I don´t think this has anything to do with the PWR pin(s). The driver board switches are set correctly aswell, so apart from the API key there is still another problem.

lmarzen, 2024-02-11:
Sounds like we are making progress. Please share the new terminal monitor output.

Smutssch, 2024-02-11:
Just realized it could still be the PWR pins, but they are placed correctly, although I had to cut off the adapter that goes into the board because the one that came with the battery was too big, and now the end of the cables look kinda broken, so I might just have to replace the whole battery. But anyway, here is the output:

 *  Executing task: C:\Users\samue\.platformio\penv\Scripts\platformio.exe device monitor 

--- Terminal on COM3 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Battery voltage: 4.06
Connecting to 'Koti'........................................................................................................................................................................................................
Could not connect to 'Koti'
WiFi Connection Failed
Busy Timeout!
_PowerOn : 10000143
Busy Timeout!
_Update_Full : 10000111
Busy Timeout!
_PowerOff : 10000107
Failed to obtain time before deep-sleep, referencing older time.
Awake for 45.987s
Deep-sleep for 21564s

lmarzen, 2024-02-11:
Hmm so it's still not displaying anything?

Looks like your wifi credentials are incorrect now.

The battery is optional. It should work fine just plugged in with usb. So that should be sufficient for testing.

Could you please attach a photo of your wiring and additionally copy the pin configuration for your config.cpp?

Smutssch, 2024-02-11:
// PINS
// ADC pin used to measure battery voltage
const uint8_t PIN_BAT_ADC  = A2; // A0 for micro-usb firebeetle
// Pins for Waveshare e-paper Driver Board
const uint8_t PIN_EPD_BUSY = 5; // 5 for micro-usb firebeetle
const uint8_t PIN_EPD_CS   =  2;
const uint8_t PIN_EPD_RST  = 21;
const uint8_t PIN_EPD_DC   = 22;
const uint8_t PIN_EPD_SCK  = 18;
const uint8_t PIN_EPD_MISO = 19; // 19 Master-In Slave-Out not used, as no data from display
const uint8_t PIN_EPD_MOSI = 23;
const uint8_t PIN_EPD_PWR  = 14; // Irrelevant if directly connected to 3.3V
// I2C Pins used for BME280
const uint8_t PIN_BME_SDA = 17;
const uint8_t PIN_BME_SCL = 16;
const uint8_t PIN_BME_PWR =  4;   // Irrelevant if directly connected to 3.3V
const uint8_t BME_ADDRESS = 0x76; // If sensor does not work, try 0x77

[image]

[image]

Smutssch, 2024-02-11:
ADC pin was wrongly selected, just fixed it and it shows "Battery voltage: 0.00
Critically low battery voltage!"

lmarzen, 2024-02-11:
Ah. Looks like you have a FireBeetle 2 ESP32-E no the FireBeetle ESP32  (non-E). The non-E version has a micro-usb connection, while the ESP32-E version has usb c.

Correct your busy line to
```C
const uint8_t PIN_EPD_BUSY = 13; // 5 for micro-usb firebeetle
```

lmarzen, 2024-02-11:
Your battery was set correctly originally as A2 since you have a firebeetle esp32-E usb-c.
```
const uint8_t PIN_BAT_ADC  = A2; // A0 for micro-usb firebeetle
```

Smutssch, 2024-02-11:
Yep, just realized that aswell, changed them. Now the Wifi connects perfectly but there is another kind of problem with the battery aside from the API key, the display still doesn´t show anything:

 *  Executing task: C:\Users\samue\.platformio\penv\Scripts\platformio.exe device monitor 

--- Terminal on COM3 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Battery voltage: 4.09
Connecting to 'Koti'..................................................................................................................................................................
IP: 192.168.2.175
Waiting for SNTP synchronization..................................
Sunday, February 11, 2024 10:24:37
Attempting HTTP Request: api.openweathermap.org/data/3.0/onecall?lat=51.227741&lon=6.773456&lang=de&units=standard&exclude=minutely&appid={API key}
  401 Unauthorized
  401 Unauthorized
  401 Unauthorized
Busy Timeout!
_PowerOn : 10001235
Busy Timeout!
_Update_Full : 10001111
Busy Timeout!
_PowerOff : 10001107
Awake for 51.833s
Deep-sleep for 293s

lmarzen, 2024-02-11:
I just spotted an issue with your wiring.

Your gnd is off by one. so are most the other pins. 3v3 is fine.

Smutssch, 2024-02-11:
Just fixed the wiring, but the display still doesn´t show anything. I´ll try to fix it myself for now, thank you for your help.

Smutssch, 2024-02-11:
Update: Got a new API key, first the wifi didn´t connect again, but now it does. Problem is, the new key is somehow unauthorized aswell.

Smutssch, 2024-02-11:
Update: Tried a few more times, now the wifi is just not connecting

Smutssch, 2024-02-11:
Update: The red light which is normally on if connected via usb-c is off

Smutssch, 2024-02-12:
Update: Wifi is connecting again, but API key is still wrong and I´m still getting a battery error
Busy Timeout!
_PowerOn : 10001235
Busy Timeout!
_Update_Full : 10001110
Busy Timeout!
_PowerOff : 10000107
Awake for 44.410s
Deep-sleep for 835s

Smutssch, 2024-02-12:
Alright. So I ran it through again, everything except the API key seems to be fine, but the display is somehow still not showing an error, which it as far as I now, should. If this is just because the API key is wrong I´ll try to fix that first.

lmarzen, 2024-02-12:
Your wiring may still be wrong. If you send pictures again I can confirm the correctness. It will draw an error on the display if the API key is incorrect, so I wouldn't worry about the api key for now.

Smutssch, 2024-02-12:
I got the API key to work, but now my BME is now found. I still want to investigate the problem with the display though, because now I don ´t know whether it is working or not, since it shoulnd´t be showing anything. Here´s the wiring though:

[image]

Smutssch, 2024-02-12:
Well then let´s solve the problem of the BME280 first, to see if it then works.

lmarzen, 2024-02-12:
The display will still draw an error even if the BME is not found.

Smutssch, 2024-02-12:
Oh, alright, well then there is still something wrong with the display.

lmarzen, 2024-02-12:
Do you have the correct panel selected in config.h?

lmarzen, 2024-02-12:
I'm running out of troubleshooting tips. Maybe your ribbon cable is not plugged in all the way?

Smutssch, 2024-02-12:
The panel is selected correctly, and the ribbon cable is plugged in all the way. Although I think the ribbon cable might be broken because there is some wrinkles in there. In that case, I would have to buy a new E-paper.

Smutssch, 2024-02-12:
Just saw this, I don´t know if that affects the performance of the e-paper, but if it does, we know the solution now.
[image]

lmarzen, 2024-02-12:
Seems cosmetic only. A wrinkled cable isn't that unordinary. Cable is probably only a problem if it is cut/sliced.

lmarzen, 2024-02-12:
You could try just connecting the driver board directly to the display without the extra ribbon cable in between.

lmarzen, 2024-02-12:
You may have already checked this but Ill say it just incase, check that the cable is in the right orientation. i.e. that it is not flipped in a way that the connections are backwards.

Smutssch, 2024-02-12:
I connected the driver board directly to the display, and it is oriented the right way, but now my wifi is not connecting again. I think the screen should show an error but it doesn´t show anything.

lmarzen, 2024-02-12:
The wifi isn't working anymore? You didn't change the ssid or password?

Smutssch, 2024-02-12:
No, I didn´t change anything in the code.

lmarzen, 2024-02-12:
Can you share the output.

Smutssch, 2024-02-12:
*  Executing task: C:\Users\samue\.platformio\penv\Scripts\platformio.exe device monitor 

--- Terminal on COM3 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Battery voltage: 4.08
Connecting to 'Koti'........................................................................................................................................................................................................
Could not connect to 'Koti'
WiFi Connection Failed
Busy Timeout!
_PowerOn : 10001143
Busy Timeout!
_Update_Full : 10001111
Busy Timeout!
_PowerOff : 10001107
Failed to obtain time before deep-sleep, referencing older time.
Awake for 46.026s
Deep-sleep for 21564s

lmarzen, 2024-02-12:
Oh looks like the connection may have timed out.

You can try inceasing the timeout in config.cpp
```c++
const unsigned long WIFI_TIMEOUT = 10000; // ms, WiFi connection timeout.
```

Sometimes if you just flashed it the esp32 can be weird so if this is the first boot after a flash, hit the reboot button and see if that resolves it.

Smutssch, 2024-02-12:
I set the timeout to 30000 now, and rebooted the esp, but still, it could not connect to my wifi.

Smutssch, 2024-02-12:
Yes, should I increase it even further then?

lmarzen, 2024-02-12:
Probably won't help. Are you far from your access point?

Smutssch, 2024-02-12:
It´s just in the next room, so no.

Smutssch, 2024-02-12:
Wifi connects again, somehow. But the display still isn´t showing anything and my BME is not found.

Battery voltage: 4.12
Connecting to 'Koti'....................................................
IP: 192.168.2.175
Waiting for SNTP synchronization..............................
Monday, February 12, 2024 20:54:30
Attempting HTTP Request: api.openweathermap.org/data/3.0/onecall?lat=51.261280&lon=6.677710&lang=de&units=standard&exclude=minutely&appid={API key}
  200 OK
Attempting HTTP Request: api.openweathermap.org/data/2.5/air_pollution/history?lat=51.261280&lon=6.677710&start=1707681273&end=1707767672&appid={API key}
  200 OK
Reading from BME280... BME not found
_PowerOn : 22
_Update_Full : 2
_PowerOff : 2
Awake for 11.726s
Deep-sleep for 214s

Smutssch, 2024-02-12:
Never mind, the wifi again isn´t connecting. At this point it just seems random if it connects or not

lmarzen, 2024-02-12:
Oh and you got your API key working. What BME are you using? Check config.cpp, for the BME address variable. Assuming there is no problem with wiring, it should work unless the address is wrong. There are 2 possible values. see the comment

Smutssch, 2024-02-12:
I´m using the BME280, I tried both 0x76 and 0x77 but it still doens´t find my BME

lmarzen, 2024-02-12:
This is unusual for sure. It sounds like it could be a hardware-related issue given the symptoms. Let's try to elimate some variables. completely disconnect the BME and Battery. Power it from usb. See if that fixes the wifi or display issue. If wifi is still misbehaving, disconnect everything. If wifi is still misbehaving then there may be an issue with your esp32 or software configuration.

lmarzen, 2024-02-12:
Also, check that you are running the software you are running is the latest version from github, last commit: Jan 12, 2024.

Smutssch, 2024-02-12:
I disconnected the battery and the BME, now the wifi works. The screen still doesn´t show anything though.

Smutssch, 2024-02-12:
Just connected the BME again, now the wifi doesn´t work, so something is wrong with the BME

Smutssch, 2024-02-12:
Alright, I don´t know how exactly but I got both the BME and the Wifi to work. But I am still getting errors:

Reading from BME280... Success
Busy Timeout!
_PowerOn : 10001246
Busy Timeout!
_Update_Full : 10001115
Busy Timeout!
_PowerOff : 10001107
Awake for 43.251s
Deep-sleep for 1397s

lmarzen, 2024-02-12:
Maybe something is shorted or wired backwards. I would go without the BME until you have the rest functioning nominally.

lmarzen, 2024-02-12:
Oh that's good progress! Im glad to see wifi is working and with the BME as well.

lmarzen, 2024-02-12:
Do you have the waveshare hat r2.2 or r2.3?

Smutssch, 2024-02-12:
I have the r2.3, but somehow the wifi is again not working. At this point it feels like it´s random whether it works or not.

lmarzen, 2024-02-12:
This project was originally developed for and only officially supports the waveshare r2.2 hat and the DESPI-C02. Some people have gotten the r2.3 working with some modifications, but I can't assist much on that front since I don't have the r2.3 hat. I recommend buying the DESPI-C02, since it is also better for low-power.

lmarzen, 2024-02-12:
I'm guessing however you have wired the r2.3 to the esp32 it is not playing well and may be drawing too much power or something that is causing your inconsistent wifi problems.

[71 more replies]