lmarzen/esp32-weather-epd #82: e-paper display palely

Mor1arty, 2024-02-16:
Hello!

I'm using waveshare E-Paper_ESP32_Driver_Board with 7.5 inch e-Paper (V2).

I think my configurations are right as the e-Paper did display contents.

However, when the driver board restarted, the e-Paper faded in and out. After that, the e-Paper remained in pale pictures and words, which could be hardly seen without strong light.

I've tried the official display demo project, and the e-Paper could show everything normally.

Could you please help me fix it?

[image]

[image]

Mor1arty, 2024-02-17:
Thanks for your help!

I'm using waveshare esp32 e-Paper Driver Board isntead of firebeetle esp32-e board.

I reset all my configurations and tighten the wires, then my problem fixed!

Thanks again!

lmarzen, 2024-02-17:
Try setting your switch to 0.47R (A).

Mor1arty, 2024-02-17:
> Try setting your switch to 0.47R (A).
> 
> It seems that for the driver hat, 0.47R is B, but on the esp32 waveshare board 0.47R is A?

Yes exactly!

The e-Paper works perfectly after I turn the switch on the esp32 waveshare board from B( by default) to A.

Thank you! You are so cool!

lmarzen, 2024-02-17:
Glad to hear that was it!

closed 2024-02-17 by lmarzen