lmarzen/esp32-weather-epd #85: 4.2 inch display version

teschmitt, 2024-03-07:
Thanks so much for all the work you've put into this project ❤️ I have started trying to squeeze as much information originally meant for the 7.5" display onto a 4.2" diagonal and it's a bit fiddly but fun.

I'd be interested in contributing to the project. Would you also be interested in a PR? If yes: How would you prefer this be implemented? I've tried incorporating yet another flag for the E-Paper panel into the renderer. While this is how it's been done for the other 7.5" displays, the smaller display requires much more customization and it gets really messy really quickly. An alternative would be to have one special renderer for each screen size or something like that. Or if you happen to have any ideas?

No sweat if you want to keep a uniform display size, btw. In that case I'll just hack together something customized for my personal use.

lmarzen, 2024-03-07:
I'm glad to hear you are interested in getting involved in the project. I would definitely work with you to get your feature merged in.

How exactly it should be implemented is a good question. The existing code was definitely not written with support for other display sizes in mind. This may require some refactoring. Maybe code that is particular to a specific display should be moved into their own functions. At least that might be my first approach, but I'll give you some freedom to decide how to handle it, and when you're ready to merge, I can refactor it some more I want.

-Luke

lmarzen, 2024-04-29:
Closed, since topic is inactive. Feel free to continue discussion of this topic here. Will reopen if there is renewed discussion.

closed 2024-04-29 by lmarzen