lmarzen/esp32-weather-epd #87: BMP280

m157, 2024-03-21:
Found out my aliexpress BME280s are actually BME280s, Is there any way to get them working on this project?

lmarzen, 2024-03-21:
Can you clarify? Provide a link or picture, please?

lmarzen, 2024-03-21:
Do you mean to say you bought BME280s and received BMP280s? There is no support currently for the BMP280.

m157, 2024-03-21:
Thanks for the response, but yeah I have BMPs, I realized I typod on the original message

lmarzen, 2024-03-21:
Sorry, there are currently no plans to add support for the BMP280. However, the display will still work fine without it attached; the only side effect of not having a sensor connected is that the indoor temperature and humidity will display dashes and a notice will appear at the bottom of the screen saying the the BME could not be detected. So you can build it and program the esp32 with the software and use it just fine until the correct sensor arrives. Hope this helps. :)

closed 2024-03-21 by lmarzen