lmarzen/esp32-weather-epd #88: Random line shifts with display refresh

Spanholz, 2024-03-22:
I built this weather station for a relative. But from time to time I get these random shifts in the upper part of the display. 

If I trigger a refresh it might disappear but may also come back in a similar way. That's why I suspect that the input data is not a problem. 

Hardware: 3-colour waveshare display with Firebeetle ESP32 ver. 1, no battery attached

[image]

lmarzen, 2024-03-22:
That's a new one 😳

Very peculiar. The images being sliced is especially weird.  Also, most of the elements misplaced, like the sunset/sunrise time have hard programmed locations where they should be drawn.

Can you confirm which driver board you are using?

I do not have the 3-color panel and can not recreate this issue on the 2-color panel.
To me, this says that there is a configuration issue, driver issue, or wiring issue.
1. Check that your driver board has its switch set to 0.47R.
2. Check that your panel is correctly in config.h. (i.e. `#define DISP_3C_B` is uncommented)
3. Double-check that your wiring matches the wiring diagram and that it matches the configuration in config.cpp.

P.S. I like the stand!

schymatter, 2024-04-07:
> I built this weather station for a relative. But from time to time I get these random shifts in the upper part of the display.
> 
> If I trigger a refresh it might disappear but may also come back in a similar way. That's why I suspect that the input data is not a problem.
> 
> Hardware: 3-colour waveshare display with Firebeetle ESP32 ver. 1, no battery attached
> 
> ![PXL_20240210_142702480](https://private-user-images.githubusercontent.com/25983258/316153616-bbc137f7-a87c-44ed-b27e-cc4aada55dfc.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI0NTc1MDksIm5iZiI6MTcxMjQ1NzIwOSwicGF0aCI6Ii8yNTk4MzI1OC8zMTYxNTM2MTYtYmJjMTM3ZjctYTg3Yy00NGVkLWIyN2UtY2M0YWFkYTU1ZGZjLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDA3VDAyMzMyOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZmZWY0MjhmOWJkYTYzNzQ1NGUzMjMwNDExZWRjMmFiZmZiZmNhNzQ1NWFkZTQ0NTc1YjJjYzE0Y2FhZTY4OGYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.JwooV5uwY5wlLQce5aNsb3q8IfOI2W81pVc_N4SkPNo)

Hello，Spanholz.I have a 3-colour waveshare display too.Would you mind share how to modify codes to satisfy 3-colour display.Thanks!

closed 2024-04-29 by lmarzen