lmarzen/esp32-weather-epd #89: No such file or directory

RBasyc, 2024-03-29:
Building in release mode
Compiling .pio\build\dfrobot_firebeetle2_esp32e\src\_strftime.cpp.o
In file included from include/_locale.h:22,
                 from src/_strftime.cpp:45:
C:/Users/Basyc/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:33:10: fatal error: freertos/FreeRTOS.h: No such file or directory   
 #include "freertos/FreeRTOS.h"
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pio\build\dfrobot_firebeetle2_esp32e\src\api_response.cpp.o
*** [.pio\build\dfrobot_firebeetle2_esp32e\src\_strftime.cpp.o] Error 1
In file included from .pio/libdeps/dfrobot_firebeetle2_esp32e/ArduinoJson/src/ArduinoJson.hpp:18,
                 from .pio/libdeps/dfrobot_firebeetle2_esp32e/ArduinoJson/src/ArduinoJson.h:9,
                 from src/api_response.cpp:19:
C:/Users/Basyc/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:33:10: fatal error: freertos/FreeRTOS.h: No such file or directory   
 #include "freertos/FreeRTOS.h"
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\dfrobot_firebeetle2_esp32e\src\api_response.cpp.o] Error 1
================================================================ [FAILED] Took 3.46 seconds ================================================================

Environment                 Status    Duration
--------------------------  --------  ------------
dfrobot_firebeetle2_esp32e  FAILED    00:00:03.463
========================================================== 1 failed, 0 succeeded in 00:00:03.463 ========================================================== 
It`s so wired

lmarzen, 2024-03-29:
Try this... https://community.platformio.org/t/freertos-freertos-h-no-such-file-or-directory/24232

RBasyc, 2024-03-29:
Hello, this problem is solved. If I don't want to use a battery pack, is it possible to realize the display by powering it only with a usb power supply?

> Try this... https://community.platformio.org/t/freertos-freertos-h-no-such-file-or-directory/24232

closed 2024-03-29 by lmarzen

RBasyc, 2024-03-30:
Can I ask if I don't have a credit card can I use the api to make the call? Or if I use another weather site's api, do I need to refactor the file that handles the response?

lmarzen, 2024-03-30:
They didn't used to require a credit card, but now they do :(  If you want to use another api it would require some refactor, yes.

RBasyc, 2024-03-30:
Thank you for the Q&A, it has helped me a lot!