lmarzen/esp32-weather-epd #90: Maybe Problem with my screen configuration

schymatter, 2024-04-05:
Dear lmarzen,

After upload successfully,My screen was lit up but turn to black seconds later,I hardly see  information on screen.How to fix it?It circulated about every 10 min.
Thanks!

lmarzen, 2024-04-05:
If the information is faint, this typically indicates low voltage. Check that the switch on your driver board is set correctly to 0.47.

To better help you, please share what driver board, panel, and esp32 board you are using.

schymatter, 2024-04-05:
> 如果信息很模糊，这通常表明面板电压较低。检查驱动板上的开关是否正确设置为 0.47。
> 
> 为了更好地帮助您，请分享您正在使用的驱动板、面板和esp32板。

[image]
I use the type-c connect to my computer as the 3.7V battery ,is it OK? Or , the 3.7v battery is involved.

lmarzen, 2024-04-05:
Type-C and/or 3.7v battery is fine.

What driver board are you using? The DESPI-C02 or the Waveshare E-Paper Driver HAT rev2.2/rev2.3?

schymatter, 2024-04-05:
> E-Paper Driver HAT rev2.2/rev2.3
E-Paper Driver HAT rev2.3

schymatter, 2024-04-05:
> Type-C and/or 3.7v battery is fine.
> 
> What driver board are you using? The DESPI-C02 or the Waveshare E-Paper Driver HAT rev2.2/rev2.3?

Now,Low battery shown on screen

lmarzen, 2024-04-05:
The rev2.3 is not officially supported by this project, instead, you should buy the DESPI-C02.

There have been similar reports of issues with low contrast using the rev2.3 #42 #62 #82. Some people have found working solutions for the rev2.3 so if you are dedicated to making the rev2.3 work, then start there.

schymatter, 2024-04-14:
Hi Imarzen，
I change the board to DESPI-C02 ,but the screen is still so weak to see.Maybe somewhere connection is not available
[image]
.Would you please give some advise,thanks!

lmarzen, 2024-04-14:
Please confirm the following:
- that the switch on your DESPI-CO2 is set correctly to 0.47
- that #define DISP_BW_V2 and #define DRIVER_DESPI_C02 are uncommented in config.h
- that you are running the latest software

If none of those things are the problem, please provide your serial monitor output.

lmarzen, 2024-04-14:
I also just noticed that your wiring for your DESPI might be incorrect. (though it is hard to tell from the blurry image)

One error is that it looks like SCK goes to 3.3V in your photo (your white wire).

[image]

schymatter, 2024-04-14:
> 我还注意到您的 DESPI 接线可能不正确。 （虽然从模糊的图像中很难看出）
> 
> 一个错误是，在你的照片中 SCK 看起来像是 3.3V（你的白线）。
> 
> 请确认您的接线与此接线图中所示的接线相符。
> 
> ![图像](https://private-user-images.githubusercontent.com/98473324/322279041-d5cc7dd7-a2f8-4267-a076-79c48c9612cb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMxMTA1NjIsIm5iZiI6MTcxMzExMDI2MiwicGF0aCI6Ii85ODQ3MzMyNC8zMjIyNzkwNDEtZDVjYzdkZDctYTJmOC00MjY3LWEwNzYtNzljNDhjOTYxMmNiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDE0VDE1NTc0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY2MjBjY2MzODVlMTEwMzgyNDM1ZWQ2YmZhMzhiNmMwZjMwZjAzOWFiZDU2YzVjMmIxNTdjMTNiOTM2NDY3ZmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.whwQ4wcYsUTV9f-WvZqgZfEPqDdhUZ92mOAYBEAPjsY)

I connect 3v3 pin and BME-pwr directly to 3v3,if not ,my screen turn to blurred after upload，is it OK?

lmarzen, 2024-04-14:
Please send a clearer picture of your wiring.

schymatter, 2024-04-14:
> 请发送更清晰的接线图片。

[image]

lmarzen, 2024-04-14:
Your wiring looks good.

How about...

> Please confirm the following:
> - that the switch on your DESPI-CO2 is set correctly to 0.47
> - that `#define DISP_BW_V2` and `#define DRIVER_DESPI_C02` are uncommented in config.h
> - that you are [...]

schymatter, 2024-04-14:
> * uncommented

After  uncommented,errors appear 
[image]

lmarzen, 2024-04-14:
Means your code should look like this...
```C
// E-PAPER PANEL
// This project supports the following E-Paper panels:
//   DISP_BW_V2 - 7.5in e-paper (v2)      800x480px  Black/White
//   DISP_3C_B  - 7.5in e-Paper (B)       800x480px  Red/Black/White
//   DISP_7C_F  - 7.3in ACeP e-Paper (F)  800x480px  7-Color
//   DISP_BW_V1 - 7.5in e-paper (v1)      640x384px  Black/White
// Uncomment the macro that identifies your physical panel.
#define DISP_BW_V2
// #define DISP_3C_B
// #define DISP_7C_F
// #define DISP_BW_V1

// E-PAPER DRIVER BOARD
// Uncomment the macro that identifies your driver board hardware.
#define DRIVER_DESPI_C02
// #define DRIVER_WAVESHARE
```

schymatter, 2024-04-14:
> 意味着你的代码应该是这样的......
> 
> ```c
> // E-PAPER PANEL
> // This project supports the following E-Paper panels:
> //   DISP_BW_V2 - 7.5in e-paper (v2)      800x480px  Black/White
> //   DISP_3C_B  - 7.5in e-Paper (B)       800x480px  Red/Black/White
> //   DISP_7C_F  - 7.3in ACeP e-Paper (F)  800x480px  7-Color
> //   DISP_BW_V1 - 7.5in e-paper (v1)      640x384px  Black/White
> // Uncomment the macro that identifies your physical panel.
> #define DISP_BW_V2
> // #define DISP_3C_B
> // #define DISP_7C_F
> // #define DISP_BW_V1
> 
> // E-PAPER DRIVER BOARD
> // Uncomment the macro that identifies your driver board hardware.
> #define DRIVER_DESPI_C02
> // #define DRIVER_WAVESHARE
> ```

Just like this before

lmarzen, 2024-04-14:
Please provide your serial monitor output.

schymatter, 2024-04-15:
> Please provide your serial monitor output.

--- Terminal on COM3 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (583) esp_core_dump_flash: Core dump data check failed:
Calculated checksum='ec239e63'
Image checksum='33333333'
Battery voltage: 4178mv
Connecting to 'HUAWEI-XB'...............................................................................................................................................................................................
IP: 192.168.1.116
Waiting for SNTP synchronization.............
Monday, April 15, 2024 11:23:43
Attempting HTTP request: api.openweathermap.org/data/3.0/onecall?lat=39.800713&lon=116.342904&lang=en&units=standard&exclude=minutely&appid={API key}
  200 OK
Attempting HTTP request: api.openweathermap.org/data/2.5/air_pollution/history?lat=39.800713&lon=116.342904&start=1713108229&end=1713194628&appid={API key}
  200 OK
Reading from BME280... Success
_PowerOn : 125996
Busy Timeout!
_Update_Full : 10001121
Busy Timeout!
_PowerOff : 10001115
Awake for 42.036s
Entering deep sleep for 356s

lmarzen, 2024-04-15:
`Busy Timeout!` is unexpected and typically indicates a wiring issue, pin configuration issue (config.cpp), or incorrect switch position on driver board.

What model of epaper panel are you using? I just assumed that it was the 7.5in e-paper (v2) 800x480px Black/White. If it is the colored version make sure the correct panel is selected in config.h.

schymatter, 2024-04-15:
> `Busy Timeout!`是意外的，通常表示接线问题、引脚配置问题 (config.cpp) 或驱动板上的开关位置不正确。
> 
> 您使用什么型号的电子纸面板？我只是假设它是 7.5 英寸电子纸 (v2) 800x480px 黑白。如果是彩色版本，请确保在 config.h 中选择了正确的面板。

 7.5in e-Paper (B)       800x480px  Red/Black/White

lmarzen, 2024-04-15:
BINGO! I should have asked that earlier.

Try this...
```C
// E-PAPER PANEL
// This project supports the following E-Paper panels:
//   DISP_BW_V2 - 7.5in e-paper (v2)      800x480px  Black/White
//   DISP_3C_B  - 7.5in e-Paper (B)       800x480px  Red/Black/White
//   DISP_7C_F  - 7.3in ACeP e-Paper (F)  800x480px  7-Color
//   DISP_BW_V1 - 7.5in e-paper (v1)      640x384px  Black/White
// Uncomment the macro that identifies your physical panel.
// #define DISP_BW_V2
#define DISP_3C_B
// #define DISP_7C_F
// #define DISP_BW_V1
```

schymatter, 2024-04-15:
> BINGO! I should have asked that earlier.
>
> Try this...
>
> ```c
> // E-PAPER PANEL
> // This project supports the following E-Paper panels:
> //   DISP_BW_V2 - 7.5in e-paper (v2)      800x480px  Black/White [...]

I got it.That's the key.Thanks a lot!

lmarzen, 2024-04-15:
Perfect! Glad to hear you got it working. Enjoy your weather display!