lmarzen/esp32-weather-epd #91: e-paper driver HAT 2.2 vs. 2.3

Nykaer, 2024-04-07:
Having an issue with the e-paper not doing anything at all.  

Found that the HAT is rev. 2.3, and there seem to be some changes to pins
https://www.waveshare.com/wiki/E-Paper_Driver_HAT

"RST pin now is for reset control, and not for power on/off".    

Could that be an issue??

lmarzen, 2024-04-07:
Yes. This project does not currently support the waveshare hat rev2.3.

Nykaer, 2024-04-07:
Thought as much..   Seems odd, that a breaking change does not bump the rev. number to at least 3.0.

lmarzen, 2024-04-07:
Yeah, same. Some people have got the rev2.3 working with this project using various work arounds. If you're curious, search the other issues on this github page. I personally don't own a rev2.3 board and haven't bothered ordering one just to debug it. The despi-c02 is a better option for low power applications, so I recommend that instead.

closed 2024-04-29 by lmarzen

ltwin8, 2024-08-03:
see #199