lmarzen/esp32-weather-epd #92: Battery Not Working?

TheMeanCanEHdian, 2024-04-11:
Hello,

I'm having a battery issue and wanted to check if there might be a code solution before I wait months to try a replacement. I have 2 Fire Beetle ESP32-E units. Both power on fine with USB. When I connect a battery the red battery light stops flashing, but there is no indication it is charging the battery (when connected via USB) or powering on (when not connected to USB).

I used a multimeter to verify the voltage coming off the pins into the Fire Beetle. Is there anything in the code that might need to be adjusted or is the functionality of the battery lower level than that?

Any help would be greatly appreciated!

lmarzen, 2024-04-11:
From the dfrobot wiki: https://wiki.dfrobot.com/FireBeetle_Board_ESP32_E_SKU_DFR0654
> Charging Indicator 	Red LED for indicating charging status: 1. Off when fully charged or not charged; 2. On when charging; 3. Quick flash when powered by USB, and no battery connected.

Measure the voltage across the positive and negative terminals of the battery, what is it?

TheMeanCanEHdian, 2024-04-11:
> From the dfrobot wiki: https://wiki.dfrobot.com/FireBeetle_Board_ESP32_E_SKU_DFR0654
>
>
> Measure the voltage across the positive and negative terminals of the battery, what is it?

You are too quick for me! I just added the link to the battery: https://www.amazon.com/gp/product/B08TTLD8VY

lmarzen, 2024-04-11:
What does the display tell you about the battery level?

Also see these options in config.h
```C
// STATUS BAR EXTRAS
//   Extra information that can be displayed on the status bar. Set to 1 to
//   enable.
#define STATUS_BAR_EXTRAS_BAT_VOLTAGE 0
#define STATUS_BAR_EXTRAS_WIFI_RSSI   0

// BATTERY MONITORING
//   You may choose to power your whether display with or without a battery.
//   Low power behavior can be controlled in config.cpp.
//   If you wish to disable battery monitoring set this macro to 0.
#define BATTERY_MONITORING 1
```

TheMeanCanEHdian, 2024-04-11:
- Battery voltage (while connected to Fire Beetle with no USB) reports 3.93 with my multimeter
- All those config.h settings are set to 1
- Display reports low battery when I connect to USB to power it on
- The red LED is not active to indicate it is charging the battery

lmarzen, 2024-04-11:
Hmm, interesting. The code relies on the factory voltage calibration to calculate voltage.
https://github.com/lmarzen/esp32-weather-epd/blob/main/platformio/src/display_utils.cpp#L42

Just asking before we proceed with further debugging, the polarity of the connectors of batteries like the one you bought are often reversed compared to what the firebeetle board connector expects. Have you checked that the connector has the correct polarity? Did you need to swap the leads in the connector?

TheMeanCanEHdian, 2024-04-11:
Oh, that's an interesting thing I didn't consider. Let me see what I can determine and report back.

TheMeanCanEHdian, 2024-04-11:
Absolutely brilliant, that was the issue! You'll be hearing from me again in the near future as I am in the process of finalizing a new style of 3D printed enclosure for this project. Thanks for creating it!

closed 2024-04-11 by TheMeanCanEHdian

lmarzen, 2024-04-11:
Glad that was it! Excited to see your enclosure design.