lmarzen/esp32-weather-epd #95: Washed out display after refresh

TheMeanCanEHdian, 2024-04-16:
Today I noticed a new problem where the display is washed out and splotchy over a certain vertical threshold. I was wonder if you had any thoughts on what might be causing this outside of a hardware issue. Below are an image showing the issue and a video showing the refresh process (after a reset).

The display looks sharp while refreshing, so I am not sure what might be the issue.

[image]

https://github.com/lmarzen/esp32-weather-epd/assets/16025103/c0df2252-cd29-479f-b8dc-800b93110626

TheMeanCanEHdian, 2024-04-16:
I also tried the following with no change:
- Reseating the display cables and connectors
- Removing and reconnecting the battery
- Powering with USB and battery connected
- Powering with only USB connected

lmarzen, 2024-04-16:
What driver board are you using (Waveshare Rev2.2/2.3 or DESPI-CO2)? Make sure the switch(es) are set as specified in the README.

TheMeanCanEHdian, 2024-04-16:
The switch was set as specified (when I move it to the other position it is way worse) but when I moved it back it now seems good? Perhaps the switch was slightly loose. It's all locked in my enclosure so not sure what would have happened but it's looking good!

Perhaps this will help someone in the future. Thanks!

closed 2024-04-16 by TheMeanCanEHdian

lmarzen, 2024-04-16:
Interesting. Thanks for responding with your fix. I didn't even think that the switch might not be all the way set in position. This may help others with similar issues.

ayh20, 2024-11-24:
FWIW - I came here to find the link to buy a new panel as mine had become "faded" after working just fine for the last year or so.  Just thought i'd search the posts to see if anyone else had had any failures and saw this post.

Did the the same as OP. Flipped the Display config switch, reset and the screen is dark... flipped it back and reset, screen is displaying again correctly !

Hope this helps someone else