lmarzen/esp32-weather-epd #97: May I ask what software is used to produce your simulation pictures?

FerrymanAnn, 2024-04-23:
As mentioned in the title, I am a beginner and I would like to know what software can be used to draw simulations like yours,Thank you.
[image]

lmarzen, 2024-04-23:
I used gimp to make that wiring diagram. gimp is just a general-purpose photo editing tool. There are many purpose built tools for making wiring schematics, and a quick google for "wiring schematic software" will bring up a bunch of discussions about available software.

FerrymanAnn, 2024-04-23:
> I used gimp to make that wiring diagram. gimp is a free and open-source general-purpose photo editing tool. There are many purpose built tools for making wiring schematics, and a quick google for "wiring schematic software" will bring up a bunch of discussions about available software.

Thank you for your reply. I will continue to study

closed 2024-04-23 by lmarzen