lmarzen/esp32-weather-epd #102: Sometimes Display work and most time No Display not work

Bunoklm, 2024-05-26:
Every wire is connected and working. I am using another ESP, the Wemos D1 Mini, and powering it over the integrated USB port. However, most of the time, the screen does nothing. The ESP starts and in 1 out of 100 cases, the display works. In the other 99 cases, the ESP32 starts, connects to Wi-Fi, asks the API for the latest data, but the screen does not refresh.

Sometimes it says that the battery is empty.

lmarzen, 2024-05-26:
Can you please paste your serial monitor output here.

Bunoklm, 2024-05-26:
*  Executing task in folder platformio: C:\Users\luis\.platformio\penv\Scripts\platformio.exe device monitor 

--- Terminal on COM5 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
E   \
QT JU WBattery voltage: 6132mv
Connecting to 'Hotspot-Z'..................................................................................
IP: 192.168.137.5
Waiting for SNTP synchronization...................
Sunday, May 26, 2024 15:57:05
Attempting HTTP request: api.openweathermap.org/data/3.0/onecall?lat=52.2093&lon=8.6008&lang=de&units=standard&exclude=minutely&appid={API key}
  200 OK
Attempting HTTP request: api.openweathermap.org/data/2.5/air_pollution/history?lat=52.2093&lon=8.6008&start=1716645429&end=1716731828&appid={API key}
  200 OK
Reading from BME280... BME not found
_PowerOn : 22
_Update_Full : 25
_PowerOff : 2
Awake for 11.285s
Entering deep sleep for 179s

lmarzen, 2024-05-26:
You may need to check your wiring. Note that some pins may have multiple uses. I cannot offer much support with the Wemos D1 Mini since I do not have one and have not tested this software on it. Have you edited the platform.ini file to use the right microcontroller controller config for the Wemos D1 Mini.

This indicates improper wiring or software configuration for both the BME and display.
```
Reading from BME280... BME not found
_PowerOn : 22
_Update_Full : 25
_PowerOff : 2
```

Bunoklm, 2024-05-26:
Yes, I don't have the sensor, but it has worked without it before. Or do I need it to update the display?

lmarzen, 2024-05-26:
It will work without the sensor

lmarzen, 2024-05-26:
If you want to tried examples focused on just testing the display, you can go to the display driver library repo found here: https://github.com/ZinggJM/GxEPD2

Bunoklm, 2024-05-26:
When I delete the Wi-Fi, the display does something but just shows grey.

Bunoklm, 2024-05-28:
I think the white cabel has a poor conetion

thanks for your help

lmarzen, 2024-05-28:
Glad you were able to solve the issue :)

closed 2024-05-29 by lmarzen