lmarzen/esp32-weather-epd #105: Problem with Waveshare Display only updating half of the screen

Juliiiian, 2024-06-05:
I updated the esp since the NTP server wasn't really working, the version before was close to a year old (130 commits behind). It works flawless on my other one but on this one I thought an update might fix it, but now I tested both drivers and I always update only the bottom half of the screen, like some sort of offset. But when the updating happens, it all goes black around the display and the bottom half. Just the middle upper part doesn't change. 

I don't think the display is broken, since before updating to a newer version it set the NTP error and the display worked.

Probably just an easy fix, but I triple checked every pin and it should work.

[image]

lmarzen, 2024-07-04:
Sorry for the delayed response. Were you able to get this resolved? I'm not sure qhat might be causing this, I haven't seen this one before.

Juliiiian, 2024-07-12:
I honestly don't know what the issue was, I downgraded to a version from 2023 Oktober or so and it worked. But there were some cable contacts not working, I soldered them back together and now it works. I think it was an issue on my end

closed 2024-07-12 by Juliiiian