lmarzen/esp32-weather-epd #106: Adaptive Temperature scale in forecast (Feature request; where to find)

LordSexy, 2024-06-05:
Hey guys, 

its not a bug report or anything, just something I would like to adjust and I dont know where to look.

My Display (800 pixel Waveshare 3 color; hat 2.3 working fine btw!)  shows the temperature and rain forecast for 36 hours (default:24).
The Y-axis for temperature starts at 5°C and ends at 30 °C and the forcast is between 12 °C and 21 °C for the next days. 

Where is the code which draws that, so I can scale it to "draw Y-axis, so temp forecast is around 10% hight to 90 %  hight of the graph"

I just want to use the space a bit more. I would also try to adapt the dotted grey lines, so they are not just at 5°C intervals, but scale to 2°C / 1°C if the graph is quiet flat for the day. If you dont have capacity to implement it, I would bodge it myself. No problems :D

But massive shoutouts to @lmarzen and anyone else involved. A dork like me could build it and change the first few things by myself. Cheers

lmarzen, 2024-06-05:
The interval scales automatically in 5-degree increments. So if there is a lot of temperature variation it will "zoom out" to 10 or 15 degree increments automatically etc.

Here is how to do what you are asking... (ps. I haven't tested this myself but this should hopefully be correct)

Changing 5 ->10 in these lines will change the minimum increment major tick distance from 5 to 10.
https://github.com/lmarzen/esp32-weather-epd/blob/430c9a3ac491e37220d670f85af505a2c7c25f41/platformio/src/renderer.cpp#L910
https://github.com/lmarzen/esp32-weather-epd/blob/430c9a3ac491e37220d670f85af505a2c7c25f41/platformio/src/renderer.cpp#L926

Changing 5 ->10 in these lines will change the default increment from 5 degrees to 10 degrees.
https://github.com/lmarzen/esp32-weather-epd/blob/430c9a3ac491e37220d670f85af505a2c7c25f41/platformio/src/renderer.cpp#L956

Changing the plus/minus 1's in these lines to 5 will make sure there is at least a distance of 5 from the top or bottom of the graph. You could do some math with some variables here to get the 10% that you desire every time.
https://github.com/lmarzen/esp32-weather-epd/blob/430c9a3ac491e37220d670f85af505a2c7c25f41/platformio/src/renderer.cpp#L948-L961

LordSexy, 2024-06-06:
[image]
AAAAH. This makes sense. 

I was wondering, because I had 10 °C to the bottom and top.

> sure there is at least a distance of 5 from the top or bottom of the graph

That is a good idea, but yesterday it failed on me. For my taste its a bit too much and Ill see how I change it. Maybe I scale it to +-10% or Ill slap 5 °C on each side and leave the "at least" out of the equation.

Ill come back and report on my findings.

closed 2025-01-06 by lmarzen