lmarzen/esp32-weather-epd #111: Display not turning on

AmicableTarsier, 2024-06-24:
I walked through all of issue #81 and it still will not turn on. Here is the output from my serial monitor(masked lat and lon):

IP: 192.168.1.14
Waiting for SNTP synchronization......
Monday, June 24, 2024 19:04:32
Attempting HTTP request: api.openweathermap.org/data/3.0/onecall?lat=lat&lon=-lon&lang=en&units=standard&exclude=minutely&appid={API key}
  200 OK
Attempting HTTP request: api.openweathermap.org/data/2.5/air_pollution/history?lat=lat&lon=-lon&start=1719183876&end=1719270275&appid={API key}
  200 OK
Reading from BME280... BME not found
_PowerOn : 22
_Update_Full : 1001
_PowerOff : 2
Awake for 7.806s
Entering deep sleep for 1532s

Also not sure why the BME280 isn't working. I've unplugged and replugged everything, fixed the polarity of my battery and still nothing. Not sure what it could be.

lmarzen, 2024-06-24:
It appears to be a wiring issue or pin configuration issue. It's not detecting the display or bme.

Please let me know what model of esp32 and what model of driver board you are using.

I can confirm the correctness of your wiring if you provide a photo. Also, copy and paste your pin configuration from config.cpp.

AmicableTarsier, 2024-06-24:
Here are some photos of everything plugged in. I’ve realized having the same color wires is not ideal, but they’re what I’ve got
[image]
[image]
[image]
[image]

AmicableTarsier, 2024-06-24:
> It appears to be a wiring issue or pin configuration issue. It's not detecting the display or bme.
>
> Please let me know what model of esp32 and what model of driver board you are using.
>
> I can [...]

firebeetle esp32e

`// ADC pin used to measure battery voltage
const uint8_t PIN_BAT_ADC  = A2; // A0 for micro-usb firebeetle
// Pins for Waveshare e-paper Driver Board
const uint8_t PIN_EPD_BUSY = 14; // 5 for micro-usb firebeetle
const uint8_t PIN_EPD_CS   = 13;
const uint8_t PIN_EPD_RST  = 21;
const uint8_t PIN_EPD_DC   = 22;
const uint8_t PIN_EPD_SCK  = 18;
const uint8_t PIN_EPD_MISO = 19; // 19 Master-In Slave-Out not used, as no data from display
const uint8_t PIN_EPD_MOSI = 23;
const uint8_t PIN_EPD_PWR  = 26; // Irrelevant if directly connected to 3.3V
// I2C Pins used for BME280
const uint8_t PIN_BME_SDA = 17;
const uint8_t PIN_BME_SCL = 16;
const uint8_t PIN_BME_PWR =  4;   // Irrelevant if directly connected to 3.3V
const uint8_t BME_ADDRESS = 0x76; `

lmarzen, 2024-06-24:
The wires between your esp32 and despi look to be wired correctly. Try moving the switch on the despi from 0.47 to 3 and back, in the past some people have reported this fixed their issue saying that the switch wasn't all the way in position.

you can try changing the BME280 address to 0x77. If you can find the data sheet for your exact model it might tell you what address is correct.
```C
const uint8_t BME_ADDRESS = 0x77;
```

AmicableTarsier, 2024-06-24:
Tried those fixes and still nothing. I've reflashed multiple times(including now) and here's the output:

Reading from BME280... BME not found
Busy Timeout!
_PowerOn : 10001242
Busy Timeout!
_Update_Full : 10000135
Busy Timeout!
_PowerOff : 10000123
Awake for 39.577s
Entering deep sleep for 1174s

I've also tried it in the 3 position, switching back forth a few times, trying it with the 3.3v connected directly to 3.3v pinout and still nothing. Could the board be defective? This is my first time working with display screens like this.

Further(maybe dumb) question: Do I need to solder the pins? or will the through hole plugin via wire be enough? It seems to write to it no problem, but not sure if that's just good feelings about it or not

lmarzen, 2024-06-25:
I think your last question is the answer. Yes, you need to solder the pins to have proper electrical connections.

AmicableTarsier, 2024-06-25:
Well, I’ll do that in a bit and see if that solves all my issues.

AmicableTarsier, 2024-06-25:
Yup, solder was the problem! Thanks, that's good to know for the future

closed 2024-06-25 by AmicableTarsier